riscv/riscv-unified-db #12: Add RISC-V pseudo-instructions

james-ball-qualcomm, 2024-08-12:
Can we add defined pseudo-instructions to riscv-unified-db?

Here's a port of a post by Ved ([link](https://lists.riscv.org/g/tech-privileged/message/2090)):
"ARC discussed a question about the location of pseudo- instructions with respect to the ISA manual and concluded that a list of pseudo-instructions should be included in the ISA manual. A PR is planned."

dhower-qc, 2024-08-14:
There is a field for adding pseudo instructions to an instruction definition. For example, in `add.uw.yaml` from the B extension:

https://github.com/riscv-software-src/riscv-unified-db/blob/1ad462fa6166b6c7206ceb8180a5f8ca28aed1be/arch/inst/B/add.uw.yaml#L25-L27

It's not super thought-out, though, and could probably use some tweaking. Getting some more examples of pseudo instructions we want to define would help.

james-ball-qualcomm, 2024-08-14:
Maybe we can get a full list from someone with experience with the RISC-V toolchains (like Ana)?

dhower-qc, 2024-08-15:
Here's an interesting case:

RDCYCLE is a pseudo instruction defined by Zicntr, but it's an alias of `csrrs` from Zicsr. The current schema that adds the pseudo instruction under the real instruction isn't a great match since `csrrs` says it belongs to Zicsr, not Zicntr. 

A few possibilities:

1. The existing pseudoinstructions field can add a `definedBy` key, similar to how CSR fields can have their own `definedBy` that is different than the parent CSR.
1. Pseudo-instructions could get their own definition file.

I'm inclined to prefer 1. since it matches what we're doing for CSRs/CSR fields, but that's not a strong preference.

james-ball-qualcomm, 2024-08-15:
Either solution looks good to me.

AFOliveira, 2024-08-20:
> RDCYCLE is a pseudo instruction defined by Zicntr, but it's an alias of `csrrs` from Zicsr. The current schema that adds the pseudo instruction under the real instruction isn't a great match since `csrrs` says it belongs to Zicsr, not Zicntr.

Please correct me if I misunderstood all of this, but doesn't this mean that the riscv-opcodes is so far wrong in defining this pseudo-instruction(and many more from what I have checked)? Since the instruction is defined as 

> $pseudo_op rv_zicsr::csrrs  rdcycle    rd 19..15=0 31..20=0xC00 14..12=2 6..2=0x1C 1..0=3

inside the **[rv_zicsr](https://github.com/riscv/riscv-opcodes/blob/master/rv_zicsr)** file. 

It seems even a bit more tricky than this since there isn't even a **rv_zicntr** file, while the spec definitely has them as two different extensions:
[image]

dhower-qc, 2024-08-20:
> Please correct me if I misunderstood all of this, but doesn't this mean that the riscv-opcodes is so far wrong in defining this pseudo-instruction(and many more from what I have checked)? Since the [...]

Yes, I believe riscv-opcodes is incorrect in this case. In general, I've found that there is quite a bit related to Zicsr/Zicntr/Zihpm that is not correct in the spec/riscv-opcodes, etc. Presumably, this is because those extensions were bolted on as an afterthought; originally, everything was defined in the base architecture.

sampsyo, 2024-10-18:
Maybe I am missing something obvious, but it seems like another wrinkle with the current format is that it's not clear what to do with pseudoinstructions that correspond to a sequence of two real instructions. For example, `call` translates to a sequence of `auipc` and `jalr`. So if pseudoinstructions don't get their own file, it is unclear which file `call` should go in (`auipc.yaml` or `jalr.yaml`?).

Perhaps this is an argument for pseudoinstructions getting their own, independent files?

AFOliveira, 2024-10-18:
@sampsyo Great question!  On top of that , sext.b case, per example would add even a little more complexity to it, since it can or can not be a single instruction.
[image]

Maybe there is no need for that and back-ends could do that if pseudoinstructions mention those parameters, but @dhower-qc or @james-ball-qualcomm  should have a better understanding of whether or not that's possible. 

Those parameters could be:

pseudoinstruction:
-AlsoDefinedBy: OtherInstruction
-doesNotExistWhen: ExtensionIsActivated 

Obviously my naming is awful :) , but I think the point remains.

dhower-qc, 2024-10-18:
This is also complicated by the fact that some multi-instruction pseduo-instructions don't have a single mapping (e.g., `li`). 

To sum up complications we've uncovered so far:

- Some pseudo-instructions defined in extension A are actually aliases of an instruction in extension B
- Some pseudo-instructions expand to a sequence of multiple instructions
- Some pseudo-instructions are replaced by a real instruction when an extension is implemented
- Some pseudo-instructions expand to one of many mulit-instruction sequences, which could be toolchain or data-dependent

Given this list, I agree with @sampsyo -- we should start to specify pseudo-instructions in their own file. The question now is how to structure it to capture these complications.

AFOliveira, 2024-10-18:
Maybe a simple way would be to use the same logic we're using now, but to specify 'original_instructions:' in pseudo-instructions files.