riscv/riscv-unified-db #24: `ldu` instruction definition is missing

ThinkOpenly, 2024-09-17:
I see `lb`, `lh`, `lw`, `ld`, `lbu`, `lhu`, `lwu`, but no `ldu`. :-)

dhower-qc, 2024-09-26:
I haven't added any part of RV128 to the database. I was under the impression that RV128 was never ratified. If that's not correct then, that's we need this database ;)

ThinkOpenly, 2024-09-26:
Ah! `ldu` is indeed RV128. I did not realize that.

Interestingly, RV128I does appear in the latest ISA spec. It is called "draft" in the Preface, but not in the (extremely terse) chapter in which it is described (Chapter 5, "RV128I Base Integer Instruction Set, Version 1.7"). It's a bit odd to me to have version > 1.0 for a draft, but it is not an impossibility. (Still a bit strange.)

Also, I can find no reference to RV128I in any of the content for content under consideration for addition to the ISA, ratified or not. Weird.

dhower-qc, 2024-09-30:
Agreed. I'd love to see the unratified stuff removed from the spec.