riscv/riscv-unified-db #44: Keep MC-1 PDF and HTML built when changes merged into main branch

james-ball-qualcomm, 2024-09-26:
I'd like to be able to point people to the MC-1 PDF and HTML files like we can for https://riscv-software-src.github.io/riscv-unified-db/html/generic_rv64/landing.html

Hold on, that link gives me a 404 error. Did the link change?

james-ball-qualcomm, 2024-10-02:
How do I send out a link to the CRD being built on the main branch (PDF & HTML) to other members of the Microcontroller WG?

dhower-qc, 2024-10-02:
> How do I send out a link to the CRD being built on the main branch (PDF & HTML) to other members of the Microcontroller WG?

https://riscv-software-src.github.io/riscv-unified-db/pdfs/MC-1.pdf

james-ball-qualcomm, 2024-10-02:
Thanks. What about the HTML? I tried https://riscv-software-src.github.io/riscv-unified-db/html/MC-1.html but nothing is there.

dhower-qc, 2024-10-02:
> Thanks. What about the HTML? I tried https://riscv-software-src.github.io/riscv-unified-db/html/MC-1.html but nothing is there.

The HTML isn't currently auto-generated

james-ball-qualcomm, 2024-10-02:
How do we auto-generate the HTML then? I thought I added that to the regression tests.

dhower-qc, 2024-10-02:
The regression test is separate from the workflow that runs when the main branch is updated. 

`.github/workflows/regress.yml`  is run on any change to a pull request. It checks that commands complete, but does not keep any artifacts.

`.github/workflows/pages.yml` is run on any update to the main branch. It generates artifacts and deploys them to GitHub pages.

closed 2024-10-15 by dhower-qc