riscv/riscv-unified-db #52: Rename XLEN to MXLEN

james-ball-qualcomm, 2024-09-30:


AFOliveira, 2024-10-04:
Hi! I know this is assigned to @dhower-qc but if this is just literally switching the names all throughout the repo? If so I can help and do it.

dhower-qc, 2024-10-04:
Excellent! It *may* be a bit more complicated than just a find/replace since XLEN is a foundational parameter; it shows up, for example, in IDL type checking. However, it might also be super simple.

I'd love for you to try and just let me know if anything strange happens.

AFOliveira, 2024-10-04:
I'll try it and follow up here!

dhower-qc, 2024-10-07:
> Just to clarify the full goal here, should only "XLEN" turn into "MXLEN" or should cases like "VSXLEN" also turn into"VS**M**XLEN"?

Just XLEN -> MXLEN

dhower-qc, 2024-11-22:
closing; will be addressed when we integrate official parameter list

dhower-qc, 2025-03-20:
This needs to be implemented now to avoid ambiguity with an "XLEN" CSR length. See #519

PR #575 (2025-04-15): Rename Sm#XLEN parameter -> Sm#MXLEN
closes #52
closed 2025-04-15 by dhower-qc