riscv/riscv-unified-db #61: Get error when adding CSR Field summary and other content from csr_adoc.erb into CRD CSR appendix

james-ball-qualcomm, 2024-10-03:
Tried adding the following and got error about mismatched []

```

==== Field Summary

[%autowidth,float="center",align="center",cols="^,<,<,<",options="header",role="stretch"]
|===
|Name | Location | Type | Reset Value

<%- csr.implemented_fields(arch_def).each do |field| -%>
| `xref:<%=csr.name%>-<%=field.name%>-def[<%= field.name %>]`
| <%= field.location_pretty(arch_def) %>
| <%= field.type(arch_def) %>
| <%= field.reset_value(arch_def) %>

<%- end -%>
|===

==== Fields

<%- if csr.implemented_fields(arch_def).empty? -%>
This CSR has no fields. However, it must still exist (not cause an `Illegal Instruction` trap) and always return zero on a read.
<%- else -%>

<%- csr.implemented_fields(arch_def).each do |field| -%>
[[<%=csr.name%>-<%=field.name%>-def]]
===== `<%= field.name %>`

[.csr-field-info]
--
Location::
`<%=field.csr.name%>[<%= field.location_pretty(arch_def) %>]`

Description::
<%= arch_def.render_erb(field.description).gsub("\n\n", "\n+\n") %>

Type::
[%autowidth]
|===

| <%= field.type(arch_def) %> | <%= field.type_desc(arch_def) %>
|===

Reset value::
<%= field.reset_value(arch_def) %>

<%- if field.has_custom_sw_write? -%>
Software write::
This field has special behavior when written by software (_e.g._, through `csrrw`).
+
When software tries to write `csr_value`, the field will be written with the return value of the function below.
+
<%- if arch_def.multi_xlen? && csr.defined_in_all_bases? && field.defined_in_all_bases? -%>
[tabs]
======
RV32::
+
[source,idl,subs="specialchars,macros"]
----
<%= field.pruned_sw_write_ast(arch_def, 32).gen_adoc %>
----

RV64::
+
[source,idl,subs="specialchars,macros"]
----
<%= field.pruned_sw_write_ast(arch_def, 64).gen_adoc %>
----
<%- else -%>
<%- xlen = !arch_def.multi_xlen? ? arch_def.mxlen : (!csr.defined_in_all_bases? ? csr.base : field.base) -%>
[source,idl,subs="specialchars,macros"]
----
<%= field.pruned_sw_write_ast(arch_def, xlen).gen_adoc %>
----
<%- end -%>
<%- end -%>

--

<%- end -%>
<%- end -%>

<%- if csr.has_custom_sw_read? -%>
==== Software read

This CSR may return a value that is different from what is stored in hardware.

[tabs]
======
Pruned::
+
[source,idl,subs="specialchars,macros"]
----
<%= csr.pruned_sw_read_ast(arch_def).gen_adoc %>
----

Original::
+
[source,idl,subs="specialchars,macros"]
----
<%= csr.type_checked_sw_read_ast(arch_def.sym_table).gen_adoc %>
----
======

<%- end # do csrs -%>`
```

dhower-qc, 2024-10-03:
Can you post the backtrace, too?

james-ball-qualcomm, 2024-10-03:
Sure, as soon as https://github.com/riscv-software-src/riscv-unified-db/issues/65 is fixed (I can't build anymore).

dhower-qc, 2024-10-15:
@james-ball-qualcomm is this fixed now?

closed 2024-10-18 by dhower-qc