riscv/riscv-unified-db #62: Field summary in Andora HTML view is messed up in some cases

james-ball-qualcomm, 2024-10-03:
[image]

dhower-qc, 2024-10-03:
The top one is just because the line is too long to fit. We'd have to come up with another way to display it to fix.

The bottom one is because the code isn't being passed through resolve_links.

AFOliveira, 2025-03-18:
Currently this table looks like this, its not super "pretty", but I think the errors in this issue have been fixed, right?

[image]

ThinkOpenly, 2025-07-27:
Shall we close this?
Alternatively, we should crisply define what the current issue is.
Note that I just opened 2 issues in this area: #927, #928.

AFOliveira, 2025-07-28:
> Shall we close this? Alternatively, we should crisply define what the current issue is. Note that I just opened 2 issues in this area: [#927](https://github.com/riscv-software-src/riscv-unified-db/issues/927), [#928](https://github.com/riscv-software-src/riscv-unified-db/issues/928).

It makes sense to me.

closed 2025-07-30 by ThinkOpenly