riscv/riscv-unified-db #63: mtval.VALUE has Asciidoc showing up in Andora HTML view

james-ball-qualcomm, 2024-10-03:
[image]

ThinkOpenly, 2025-07-27:
Shall we close this? The AsciiDoc text is gone, although there are some Ruby conditionals still bleeding through, and the formatting is sometimes sub-optimal. Examples:

No need for a bullet here:
[image]

Probably need bullets here:
[image]

Why are these broken up into 16-bit segments?
[image]

Conditional bleed-thru:
[image]

Given the interest for "good first issue", shall we close this one and open 4 new issues for the above things?

ThinkOpenly, 2025-07-27:
I opened issues for the first 2. The last two might be intentional bleed-thru? They certainly don't read well, but I'm not sure how to better format that content.