riscv/riscv-unified-db #65: While adding Smpmp extension, somehow PHYS_ADDR_WIDTH is undefined

james-ball-qualcomm, 2024-10-03:
I looked at the changes I made to Sm.yaml (just removed PMP parameters) and the creation of the new Smpmp.yaml and don't see any obvious reason why PHYS_ADDR_WIDTH would now be undefined. 

```
Generating arch/csr/I/pmpcfg8.yaml
Generating arch/csr/I/pmpcfg9.yaml

In file /github/riscv-unified-db/arch/isa/globals.isa
On line 186
In the code:

  184: 
  185: 
  186: struct TranslationResult {
  187:   Bits<PHYS_ADDR_WIDTH> paddr;
  188:   Pbmt pbmt;
  189:   PteFlags pte_flags;
  190: }

A type error occured
  no symbol named 'PHYS_ADDR_WIDTH' on line 186
/github/riscv-unified-db/lib/idl/ast.rb:279:in `type_error'
/github/riscv-unified-db/lib/idl/ast.rb:522:in `type_check'

```

james-ball-qualcomm, 2024-10-03:
Bug was found by Derek. When I copied Sm.yaml to make Smpmp.yaml, I forgot to rename the key from Sm to Smpmp. So, there were 2 Sm extensions as far as the db is concerned and instead of giving me a nice error message, it replaced the contents of one with the other or something like that.

Derek, do you want to add a nice error message?

dhower-qc, 2024-10-04:
bug fixed in #68. Created new issue #73 to track the better error message

closed 2024-10-15 by dhower-qc