riscv/riscv-unified-db #69: Zihpm.yaml missing several parameters and has vague description for HPM_COUNTER_EN

james-ball-qualcomm, 2024-10-04:
Couple of issues I noticed with Zihpm.

The Unpriv ISA manual chapter on Zihpm says "The implemented number and width of these additional counters, and the set of events they count, is platform-specific. Accessing an unimplemented or ill-configured counter may cause an illegal-instruction exception or may return a constant value."

Seems like we need a bunch more of parameters in the Zihpm.yaml to capture all these possibilities. Should we add them?

The other issue is that the Zihpm.yaml has this description for HPM_COUNTER_EN:
        "List of HPM counters that are enabled."
What does "enabled" mean? Counting some event (i.e., run-time controlled) or provides a non-zero value when the associated CSR is read (since all 29 perf counters are always implemented but can be read-only-0 if not actually functional).

dhower-qc, 2024-10-04:
Yes, I agree we are probably missing parameters for Zihpm.

As for "enabled," that means that the corresponding CSR is present (read won't cause exception or undefined behavior). It may still be read-only-0 even if "enabled".

ThinkOpenly, 2025-07-27:
https://github.com/riscv/riscv-isa-manual/issues/2160

Sukuna0007Abhi, 2025-08-03:
I think I can work on this for now, pls assign me @ThinkOpenly @dhower-qc

ThinkOpenly, 2025-08-04:
> The simplest (and most logical) implementation I think would be:
> 
> * up to 29 additional registers
> * all identical widths
> * valid widths are 32 or 64 bits
> 
> So, this issue needs configuration parameters for:
> 
> * how many additional `hpmevent` registers (up to 29)
> * register width (32 or 64)
> * behavior for access to unimplemented register: illegal instruction or some constant value
>   
>   * what is the constant value

There were a few responses in https://github.com/riscv/riscv-isa-manual/issues/2160, although not conclusive (to me). Here's my current thinking:
- up to 29 additional registers
- all 64 bits

So, configuration parameters for:
> * how many additional `hpmevent` registers (up to 29)
> * behavior for access to unimplemented register: illegal instruction or some constant value (and what constant value)