riscv/riscv-unified-db #77: Instruction Index Grouping

AFOliveira, 2024-10-07:
I was just now thinking of how it would look to have all instructions ordered just alphabetically and I think it will be confusing when trying to see which instruction matches what extension. I think we could solve this in two ways:

- Group instructions by extension on index.
- Create a list on the end of the extension refering to link all the instructions the extension adds.

I'm more inclined to prefer the second since the first would mean duplicated instructions

dhower-qc, 2024-10-07:
Good suggestion. We do this for some of the generated docs already (e.g., the config-specific spec). I didn't add it to the ISA manual generation because I wasn't sure how to best fit it in. In theory, the extensions are being defined by the ISA manual prose that already in there, so it seemed strange to me (at the time) to redefine them again in an appendix. But maybe that isn't so strange.

AFOliveira, 2024-10-07:
I believe that one of the main issues to point on the ISA manual is that there is no simple way to see what instructions each extension feats. Therefore, I believe this to be one of the main points where this documentation should offer a delta on. Maybe even just a simple list - even if it is horizontal just as "Instructions : ADD, SUB ..." would be really helpful - but there probably are better ways to do this that I can't think as of now.

james-ball-qualcomm, 2024-10-07:
Since we are generating the documentation from a database, we can include multiple views of the same information. For example, have an appendix of all the CSRs in alphabetical order but am about to add another appendix with them in numerical order. This is for the CRD but could also be for other documents (e.g. Profile, ISA manuals). Maybe you use this philosophy to solve the same fundamental issue in this GitHub issue: namely that people will want to see different views of the database.

AFOliveira, 2024-10-07:
I think that makes a lot of sense. I believe that this could even be handled with flags so that it would be possible to generate those different views on demand.

dhower-qc, 2024-10-16:
I've added this in #101