riscv/riscv-unified-db #79: Offline Visualization

AFOliveira, 2024-10-07:
As of now all images are hosted in kroki.io, is there any specific reason for that. I do realize this is low priority, but for future purposes I believe that we should try to avoid to rely on this external tools, right? 
Offline visualization:
[image]

dhower-qc, 2024-10-07:
Yes, we should. The only reason we're using that is because the only public plugin for wavedrom in Asciidoctor.js (needed by Antora) uses Kroki. To fix this, we'll need to write a custom plugin.

dhower-qc, 2024-10-07:
We are using wavedrom. The issue is that there are multiple asciidoctor plugins for wavedrom that can only be used with certain tools.

riscv-isa-manual uses the Ruby backend ('asciidoctor'), and the wavedrom plugin for that generates and stores wavedrom images locally. riscv-unified-db also uses the Ruby backend to create PDFs. 

However, the tool being used to create the html ISA manual out riscv-unified-db is Antora, which can only use a JavaScript backend ('asciidoctor-js') of Asciidoctor. The wavedrom plugin for asciidoctor-js is what generates those on-demand image links using Kroki. (For what it's worth, Bill Traynor from RVI has an Antora version of riscv-isa-manual, and it has the same on-demand image issue).

AFOliveira, 2024-10-07:
Got it! Thanks for the explanation!

dhower-qc, 2024-10-30:
@drom, are you aware of any alternatives that will help get static wavedrom images in pages generated from Antora? Right now, we are using the Kroki plugin that generates the image at page load time instead of page creation time.

drom, 2024-10-30:
I never used Antoria. Does it allow running some JS inside the page? we can render diagrams dynamically.