riscv/riscv-unified-db #81: Add many missing CSRs found by Breker

james-ball-qualcomm, 2024-10-08:
From Adnan:

We found the CSRs listed below to be mentioned in the privileged spec but missing in the riscv-unified-db git repository.

tselect
tdata1
tdata2
tdata3
mcontext
dcsr
dpc
dscratch0
dscratch1
fflags
frm
hideleg
hgeie
htval
hip
htinst
hgeip
henvcfgh
hcontext
hstateen0
hstateen1
hstateen2
hstateen3
hstateen0h
hstateen1h
hstateen2h
hstateen3h
vsie
vsscratch
vsip
mseccfgh
scountinhibit
scountovf
scontext
sstateen0
sstateen1
sstateen2
sstateen3
mtinst
mtval2
mstateen0
mstateen1
mstateen2
mstateen3
mstateen0h
mstateen1h
mstateen2h
mstateen3h
mnscratch
mnepc
mncause
mnstatus

dhower-qc, 2024-10-18:
This is related to generating a tracking table (#118)

AFOliveira, 2025-03-18:
@dhower-qc is this really duplicated? We are still missing a lot of those CSRs

ThinkOpenly, 2025-03-18:
> [@dhower-qc](https://github.com/dhower-qc) is this really duplicated? We are still missing a lot of those CSRs

I don't think Derek's comment from October 2024 was implying this issue is a duplicate of #118, if that's your question.

At the time, I suspect the hope with #118 and #207, to my understanding, was to track all of the missing pieces like reported here. I don't think we really have that, but a robust survey might be a good task for a mentee?

neverlandiz, 2025-03-18:
Sounds interesting, could I get some more information about this issue?

AFOliveira, 2025-03-18:
Hi @neverlandiz, I may give you my overview, but likely @james-ball-qualcomm will be the best person to help on this. RISC-V Unified DB has CSRs defined under arch/csr and there yaml files have a bunch of yaml fields defining each CSR, such as 

```
# yaml-language-server: $schema=../../schemas/csr_schema.json

$schema: "csr_schema.json#"
kind: csr
name: cycle
long_name: Cycle counter for RDCYCLE Instruction
address: 0xC00
description: |
  Alias for M-mode CSR `mcycle`.

  Privilege mode access is controlled with `mcounteren.CY`, `scounteren.CY`, and `hcounteren.CY` as follows:

  [%autowidth,cols="1,1,1,1,1,1,1",separator="!"]
  !===
  .2+h![.rotate]#`mcounteren.CY`# .2+h! [.rotate]#`scounteren.CY`# .2+h! [.rotate]#`hcounteren.CY`#
  4+^.>h! `cycle` behavior
  .^h! S-mode .^h! U-mode .^h! VS-mode .^h! VU-mode

  ! 0 ! - ! - ! `IllegalInstruction` ! `IllegalInstruction` ! `IllegalInstruction` ! `IllegalInstruction`
  ! 1 ! 0 ! 0 ! read-only ! `IllegalInstruction` ! `VirtualInstruction` ! `VirtualInstruction`
  ! 1 ! 1 ! 0 ! read-only ! read-only ! `VirtualInstruction` ! `VirtualInstruction`
  ! 1 ! 0 ! 1 ! read-only ! `IllegalInstruction` ! read-only ! `VirtualInstruction`
  ! 1 ! 1 ! 1 ! read-only ! read-only ! read-only ! read-only
  !===
priv_mode: U
length: 64
definedBy: Zicntr
fields:
  COUNT:
    location: 63-0
    alias: mcycle.COUNT
    description: Alias of `mcycle.COUNT`.
    type: RO-H
    reset_value: UNDEFINED_LEGAL
sw_read(): |
  # access is determined by *counteren CSRs
  if (mode() == PrivilegeMode::S) {
    # S-mode is present ->
    #   mcounteren determines access in S-mode
    if (CSR[mcounteren].CY == 1'b0) {
      raise(ExceptionCode::IllegalInstruction, mode(), $encoding);
```
This yaml format allways follows a json schema that you may find in /schemas/csr_schema.json

To write most of these CSRs, what you will have to do is search the Unprivileged and Privileged ISA Manual and search for the definition of each of this parameters for each CSR there - they should be explicitly described. The only field of this yamls that would need further work is _sw_read()_ because this is the IDL part of the CSR. I would say that if you are interested in tackling this, start by picking some of the CSRs that @james-ball-qualcomm listed, try to write their yaml definition and ask for revision once you have done a few, so you can get some feedback/guidance on how to lead your work. Do you want to pursue this?

ThinkOpenly, 2025-03-18:
@AFOliveira covered it well. The pertinent CSR information entirely in the Privileged manual. If it's not there, it could be in a ratified-but-not-yet-integrated extension. Those can also be found at riscv.org, underneath the table for Unpriv/Priv in a little note:
> Note: Recently ratified extensions, but not yet included in the full specifications, can be found on the RISC-V Ratified Extensions page.

That links to https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154732/Ratified+Extensions.

As Afonso suggested, try one, ask for feedback, rinse and repeat. ;-)

neverlandiz, 2025-03-19:
Thanks for the information! I'll start working on this.

closed 2025-03-27 by dhower-qc