riscv/riscv-unified-db #106: PDF generation bug

AFOliveira, 2024-10-16:
Just to keep track of this problem, there's an error in PDF generation that happens to some extensions. It was caused by a back-end update.
At least, it affects the I extension.

jalobaba, 2024-10-16:
I'm curious what the issue is. Do you have more details?

AFOliveira, 2024-10-17:
The problem happens when you try to run `./do gen:ext_pdf[I]` .
The error itself looks like this
[image]

james-ball-qualcomm, 2024-10-17:
Can you update the issue's title to make it clearer what the issue is related to?

AFOliveira, 2024-10-17:
> Can you update the issue's title to make it clearer what the issue is related to?

Surely!  Sorry if it's not clear yet, feel free to make any suggestion or change it as you think it may be better.

jalobaba, 2024-10-17:
Looks good now. Thanks!

dhower-qc, 2024-10-17:
This is caused by the fact that the extension document generator isn't being careful to check that extension data (yaml files) have all the information it's expecting. I'm working on a patch now so that it won't crash, but long term, we just want to get all the data filled in the database.

closed 2024-10-18 by dhower-qc