riscv/riscv-unified-db #99: Template helpers for links & anchors doesn't seem to work for links and no support for extension parameters

james-ball-qualcomm, 2024-10-16:
We created the template_helpers.rb file a while ago and I just noticed the CRD template is only using link_to_inst() and anchor_for_inst() and isn't using link/anchors for extensions, CSRs, and CSR fields. When I tried adding links for to extensions I just get no link. 

As a secondary issue, I added link/anchor support to the template_helpers.rb for extension parameters. The anchor looks straightforward but I'm not sure about my implementation for link because I don't understand the %%LINK% being used by the other types of links.  Why are we using %%LINK% instead of << >>?

Here's what I mean:

  # Insert a hyperlink to an extension.
  # @param name [#to_s] Name of the extension
  def link_to_ext(name)
    "%%LINK%ext;#{name};#{name}%%"
  end

  # Insert a hyperlink to an extension parameter.
  # @param ext_name [#to_s] Name of the extension
  # @param param_name [#to_s] Name of the parameter
  def link_to_ext_param(ext_name, param_name)
    "<<ext-#{ext_name.gsub(".", "_")}-param-#{param_name}-def, #{ext_name}>>"
  end

dhower-qc, 2024-10-18:
> Why are we using %%LINK% instead of << >>

Because what goes in <<>> depends on how we are putting the asciidoc together (e.g., split to fit the Antora expectations or as a single page). We use %%LINK% as a placeholder so that we can generate the right cross-reference after we know what we are generating.

ThinkOpenly, 2025-11-04:
Is this still an issue? What needs to be addressed?