riscv/riscv-unified-db #127: Additional parameter for mtvec

christian-herber-nxp, 2024-10-17:
According to the ISA manual, 

> the mtvec register must always be implemented, but can contain a read-only value

There is currently no parameter to allow mtvec.base to be read-only.
Thus, I would like to suggest adding:

```
    MTVEC_BASE_READ_ONLY:
      description: |
        Whether or not the `mtvec.base` field is read only or not.

        Possible values:

        true::
          The `mtvec.base` field is read-only.
        
        false::
          The `mtvec.base` field is read/write.
      schema:
        type: boolean
```

james-ball-qualcomm, 2024-10-17:
I had the same impression until Derek showed me something (see screenshot). The db currently assumes that if you only support one value for mtvec.base, then it is read-only. Seems like a reasonable assumption. If you agree, maybe we should just update the description to say this so that others don't get caught by this.

[image]

dhower-qc, 2024-10-17:
@james-ball-qualcomm, that's for mtvec.MODES. @christian-herber-nxp is referening to mtvec.BASE

dhower-qc, 2024-10-17:
Unfortunately, I think the allowed values are more complicated that just read-only-0 or anything since the field is WARL.

How about this?

```yaml
    MTVEC_BASE_VALUES:
      description: |
        The supported values in `mtvec.base`. 

        Possible values:

        none::
          `mtvec.base` is read-only-zero.

        width::
          `mtvec.base` supports any value that fits in MTVEC_BASE_WIDTH bits

        custom::
          `mtvec.base` supports a custom set of values. Any attempt to read or write `mtvec.BASE` will
          result in UNPREDICTABLE execution.
      schema:
        type: string
        enum: [none, width, custom]
    MTVEC_BASE_WIDTH:
      description: |
        When MTVEC_BASE_VALUES is 'width', the bit width of supported value in `mtvec.BASE`
      schema:
        type: integer
        minimum: 1
        maximum: 60
      extra_validation: |
        # max width is actually 30 for RV32
        assert MTVEC_BASE_WIDTH <= 30 if MTVEC_BASE_VALUES == "width" && XLEN == 32
```

james-ball-qualcomm, 2024-10-17:
Oops. Good point. Should we add this parameter then? Makes sense to me.

However, isn't there some general-purpose way to mark a field as read-only? This is going to come up often and I'd hate to think we have to create a parameter for each one.

dhower-qc, 2024-10-17:
> Oops. Good point. Should we add this parameter then? Makes sense to me.
> 
> However, isn't there some general-purpose way to mark a field as read-only? This is going to come up often and I'd hate to think we have to create a parameter for each one.

Maybe, but it is complicated by WARL (see my comment above)

christian-herber-nxp, 2024-10-17:
There are already parameters that limit the alignment. This makes sense.
On top, read-only is the only thing that is needed. This does not have to be read only zero.

The definition of custom above is a bit too unpredictable. There is a very predictable scenario of mtvec being hardwired to a non-zero address. Reading this is not unpredictable.

christian-herber-nxp, 2024-10-18:
> Actually, besides parameters that limit the minimum alignment (affects low-order bits) I'm assuming a valid implementation could have some of the high-order bits read-only-0 to save a few registers (if they didn't need a full 32-bit address space).

There is actually precedent around MISA. Here, there is a parameter per field to specify if it is mutable or not. So maybe mutable is the better term to use here also.

On the high order bits: Agree that this could be the case. So you would have to define a maximum value for mtvec.base, that would solve it.
I prefer a specific solution to start with. The more cases we see, the more we can generalize.
 
> I'm thinking that this issue of read-only (not necessarily 0, for example the marchid, mvendorid, mhartid, etc.) and a WARL field that can be fully described by just knowing the number of bits implemented is going to happen so often that we'll want a general-purpose facility for this and not have to create parameters manually for each one. Derek, we've talked about this in the past but I can't remember the outcome of that discussion.
> 
> Christian, BTW, the "custom" is Derek's way of saying we don't know how to represent your implementation's behavior (yet) in the db. That's handy for certification since we can say any parameter with a "custom" value is out of scope for certification or means e can't certify your implementation (yet). We'll probably add more choices to parameters over time (or add more parameters) to be able to represent more implementations but we'll be limited to some subset at any given time.

Understand. But we do not want to limit the scope of certification so far that no device passes certification because its implementation choice is out of scope.

james-ball-qualcomm, 2024-10-18:
The parameter-per MISA bit is another good example of how I'd like to replace specific parameters with a more general-purpose solution. There are already many CSR fields that would need a general-purpose mechanism so I'm not sure we want to wait for a more cases. The later we make a change, the harder it will be.

christian-herber-nxp, 2024-10-21:
I realized mtvec is actually a rather complicated scenario. The valid values in mtvec.base are actually a parameter of mtvec.mode.
Not sure how to model this efficiently and somewhat generically.

allenjbaum, 2024-10-21:
riscv-config as a schema for all this and more, including rather general
conditionals where
the legal values and formats depend on other architectural state bits .
Those could be a field in the same or other CSR, or not (e.g. the current
mode)