riscv/riscv-unified-db #130: Extension Groups

AFOliveira, 2024-10-18:
When I'm trying to generate instructions that belong to two extensions, I am getting some errors:

With this yaml:

```
c.flw:
...
  definedBy: [C, F]
...

```

I get this error:

` At /c.flw/definedBy, Expecting object, got ["C", "F"]`
`
Am I missing any step in generating the instructions or is this some type of back-end error?

dhower-qc, 2024-10-18:
Ah, sorry. The schema for definedBy has changed to make it more expressive. The new way of specifying this is:

```yaml
definedBy:
  allOf: [C, F]  # both C and F must be present (AND)
```

FYI, other boolean operations are also possible now:

```yaml
definedBy:
  anyOf: [C, F]  # either C or F must be present (OR)

definedBy:
  oneOf: [C, F] # exactly one of C or F must be present (XOR)
```

And arbitrary recursive schemas are also possible:

```yaml
definedBy:
  anyOf:
  - allOf: [C, F]
  - Zcoolext
```

AFOliveira, 2024-10-18:
Thank you for clarifying!

closed 2024-10-18 by AFOliveira