riscv/riscv-unified-db #131: PseudoInstructions defined by other instructions

AFOliveira, 2024-10-18:
How should we organize the following?

`$pseudo_op rv_zicsr::csrrs  frcsr      rd 19..15=0 31..20=0x003 14..12=2 6..2=0x1C 1..0=3
`

To those unware of the format, this means pseudoinstruction **frcsr** is a defined by instruction **csrrs** which belongs to extension **zicsr**.

Should this still only be defined in the zicsr folder? Should this have another parameter like:
```
pseudoinstructions:
  - when: (csr == 0x3) && (rs1 == 0x0)
    to: frcsr
   **ext: F**

```

AFOliveira, 2024-10-18:
I just noticed the ISA states "The F extension depends on the "Zicsr" extension for control and status register access." So this may answer my question, I guess they stay in Zicsr, right?

dhower-qc, 2024-10-18:
duplicate of #12

dhower-qc, 2024-10-18:
We clearly need to re-architect pseudo-instructions. Out of curiosity, does riscv-opcodes handle multi-instruction pseudo-instructions?

AFOliveira, 2024-10-18:
No, I was planning on fix that there, but then started focusing more on the UDB and left it hanging. That's also the reason I didn't remember this sooner, since I did not fetch that information from riscv-opcodes.

closed 2024-10-18 by AFOliveira