riscv/riscv-unified-db #132: Instructions Folder organization

AFOliveira, 2024-10-18:
When an instruction is defined by **several extensions**, i.e.   `definedBy: [B, Zbc, Zbkc, Zk, Zkn, Zks]`  , in which folder do we want the .yaml described? Possibly all of them?

dhower-qc, 2024-10-18:
We don't want to duplicate instructions in all of them -- that would cause an error since it would like an instruction is being defined multiple times. I don't have a strong preference on which sub-folder to choose. Technically, it doesn't matter; the database doesn't apply any meaning to the folders (other than the top level csr/, inst/, etc. to determine type). It's just for human organization.

AFOliveira, 2024-10-18:
Ok, then for now, I'll simply put it in the first extension that appears alphabetically.

closed 2024-10-18 by dhower-qc