riscv/riscv-unified-db #180: Add support for GitHub codespace users

james-ball-qualcomm, 2024-10-24:
For casual users of the UDB, if we had a turnkey solution for GitHub codespace users, this could aid in adoption and participation in UDB.

dhower-qc, 2024-10-24:
@wmat, do you have any experience with codespace?

trdthg, 2024-10-30:
Hi, I just learned about this project, and I am trying to study it.

I have used several projects that provide codespaces
They usually provide a `setup.sh` script, and users only need to run it; all dependencies will be automatically installed, and then it's just a matter of executing make or other things. In short, work can start directly.

This is a related PR for rustlings (a widely used RUST exercise repository for beginners): https://github.com/rust-lang/rustlings/pull/1399

wmat, 2024-10-30:
@James Ball ***@***.***> my codespace knowledge is limited.
If there is a cost for it within an org like risc-v, I'm not sure it would
be approved.

james-ball-qualcomm, 2024-10-30:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

According to GitHub's website, you get some free amount of codespace usage per month. After that, they bill the individual, not RISC-V International. I think it would be good for kicking the tires and trying out UDB. After that, if someone wanted to do more development than the free amount they either have to get their company to pay or get set up on their own computer like Derek and I have done.

+james+

dhower-qc, 2024-11-01:
This is now working.

closed 2024-11-01 by dhower-qc