riscv/riscv-unified-db #186: Revamp certificate naming and versioning scheme

james-ball-qualcomm, 2024-10-25:
I've been thinking about the certificate naming scheme and versioning scheme and how those map to the standards naming schemes and versioning scheme. This will not be 1:1 relationships because we have have many certification releases for a single standard release. As long as the database tracks the relationship between the two naming schemes (CRD and TSC standards/profiles/platforms) we should be good.

Here's a proposal:
MC100 is the most minimal microcontroller. It has no optional extensions, a few in-scope extension parameters, and many out-of-scope extension parameters. Its first release will be call MC100-v1 with an implicit 0 minor release and 0 patch release so with those included is MC100-v1.0.0. After this we might to decide to add an optional feature to MC100 such as PMP so we create MC100-v1.1.0 to indicate it is a new MC100 release with one or more optional features. Later we get enough customer demand for MC100 with a PMP but also with a CLIC. and they want them mandatory. So, instead of creating MC100-v2.0.0 we decide to create MC120-v1.0.0. Then we get need for an MC120 but with with Zicond mandatory so we create MC120-v1.1.0 (minor release). Next we create a microcontroller that has CLIC & PMP mandatory but with U-mode and S-mode optional and sPMP optional called MC200-v1.0.0. Then we want a microcontroller that is similar to MC100 but with the Zc extension mandatory so we call it MC100-v2.0.0.

jalobaba, 2024-10-25:
Let's explore apps processors. We'll focus on ratified profiles. We'll need to add more scope to them to also specify M-mode and probably debug and trace. So, first we have to decide if our certificate names will match profile names or we'll have our own names that map to theirs. If the latter we can follow the microcontroller scheme proposed and call them AC something. 

Seems like we'll have a certificate product name (e.g., MC100 or AC100) and a certificate version number (e.g., v1, v1.1, v1.2.1, v2, v3.4.5, etc.) that follows semantic numbering. What I'm not clear about is when we add new mandatory or optional features, do we update the product name (MC100 becomes MC101, MC110, or MC200) or do we keep the product name the same and update the certificate version number (e.g., v1 becomes v1.1 or v2). Seems like the former because the latter would be used as we increase allowed parameters (move from out-of-scope to in-scope).

Say AC100 is RVB23 without V and H but with M-mode and no debug, trace, PMP or MTT. This is the first apps processor we certify.

james-ball-qualcomm, 2024-10-25:
@mconley@vtmgroup.com
@shiqinghao.sqh@alibaba-inc.com

james-ball-qualcomm, 2024-10-25:
Here's a list of all the use cases I think our CRD naming/versioning will need to support. Please review this list and indicate if something isn't clearly worded, inaccurate, vague, or I'm missing a case.

1. CRD updated to improve descriptions (no functional changes)
2. Fix bugs in the certification tests. Could cause implementations that passed with the bugs in place to now fail.
3. Increase scope of certification tests to move out-of-scope parameters (listed in the CRD) to in-scope parameters or allow more values for existing in-scope parameters. We aren't adding any new extensions to the CRD; we are just testing more of the architecturally defined behaviors of CRD listed extensions. Could cause one or more certificates that passed with the original smaller scope to fail.
4. Increase scope of certification tests to test aspects of existing CRD listed extensions without any changes to their CRD listed parameters. For example, assume when we originally issued a certificate we didn't test exception priority and then added tests for that. Could cause one or more certificates that passed with the original smaller scope to fail.
5. Add certification tests for an extension previously not included in existing CRDs. Naming/versioning might want to differentiate between small extensions (e.g., Zicond) and large extensions (e.g., U-mode, PMP, H, or V). Naming/versioning might also want to differentiate between extensions added as optional vs. mandatory to CRDs.
6. TSC creates a new version of an existing standard. It could be a major release or a minor release. The TSC has versions for extensions, the Unpriv ISA manual, Priv ISA manual, Profiles, Platforms, and non-ISA standards (e.g., debug, trace, IOMMU, AIA).

james-ball-qualcomm, 2024-10-25:
@david.weaver@akeana.com

dhower-qc, 2024-10-25:
I'm going to move this to Discussions...trying to use issues to track actionable work.