riscv/riscv-unified-db #187: Instruction description enhancement

AFOliveira, 2024-10-25:
**Is your feature request related to a problem? Please describe.**
On Instructions yaml files: `Excluded by` field can only take one extensions, but an instruction could be excluded by more that one extension.

**Describe the solution you'd like**
Set it has the `definedBy:` field. Let it have anyOf, allOf.

**Additional context**
Zext.h example:
  excludedBy: [Zk, Zkn, Zks, Zbkb]

dhower-qc, 2024-10-25:
I'll get this fixed so it works just like definedBy

dhower-qc, 2024-10-25:
Actually, I just looked and I think both the schema and Ruby object model for instruction.excludedBy should accept the same as definedBy. Are you up to date with the main branch?

AFOliveira, 2024-10-25:
I think it broke my regression test, but it could have been my mistake. I'll try again and get back here.

AFOliveira, 2024-10-27:
It was my mistake. Thank you for the clarification. Already fixed `zext.h` in #188

closed 2024-10-27 by AFOliveira