riscv/riscv-unified-db #193: Need regression tests to ensure artifacts (e.g., manual, profiles, CRDs) aren't broken by changes to main

james-ball-qualcomm, 2024-10-25:
Now that we are telling people to look at the artifacts (especially the MC-1 CRD), our tolerance for breaking these artifacts is low enough that we need proper regression tests that fail if the artifacts are changed in unintentional ways. For a good example of this, see issue #192.

dhower-qc, 2024-10-25:
how do you want to check it, aside from just a successful build? Store known-good generated files?

AFOliveira, 2024-10-25:
I was thinking of proposing to add a validation check to ensure generated encodings stay consistent with riscv-opcodes. I’m working on a few final updates to make this possible, and I think this could help keep encoding tight and accurate. This would also be possible for pseudoinstructions.

james-ball-qualcomm, 2024-10-25:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Yes, storing known good artifacts and comparing against them. At least for the CRDs and the profiles we can compare the Asciidoc files that are generated. Hopefully we can have some reasonable GUI diff tool and ake it easy to update the expected file after the comparison has been done.

We can also have different kinds of regression tests depending on the branch. For example, we can have an integration branch for CRDs and Profiles that the bulk of development affecting them is done in branches off the integration branch. When a PR is performed to the integration branch the regression test doesn't bother checking the CRD and Profiles artifacts against good known ones since they changing often. However when a PR is performed to the main branch the regression test does check against the expected artifacts to prevent different users from breaking artifacts not related to their direct work. Then we can do releases off the main branch as we do now or just copy them over to GitHub pages.

+james+

james-ball-qualcomm, 2025-02-22:
We should create a mock arch directory that just contains a relatively small number of mock extensions, instruction, csrs, etc. but with a good range of different values to get coverage of the Ruby backend generator code. This allows us to control the data being fed into the generators since it would only be used for testing and not the normal data that's being used to describe the RISC-V ISA that keeps evolving and changing.

Once we have this mock arch, we could use it for unit level tests of the Ruby objects and then also could potentially compare the output adoc file that gets generated by backend generators with expected adoc without having the comparison keep breaking because people are updating the real data.

ThinkOpenly, 2025-11-04:
Can this be closed?