riscv/riscv-unified-db #198: Provide a synopsis for the RVV instructions

lu-zero, 2024-10-25:
It is currently completely missing.

dhower-qc, 2024-10-25:
Is that a volunteer request? 😃 

There are a lot of instructions that we have generated skeletons for out of riscv-opcodes. Those have no synopsis, assembly, etc. I'm going to work on a dashboard to keep track of them.

lu-zero, 2024-11-04:
Not a volunteer request yet since I do not have adequate command on the ISA to dare to document it.

I started opening it as an anchor for asm-lsp to figure out when to switch to it as source of information.

dacsson, 2025-09-13:
Is this a good area of contribution to provide synopsises to some RVV instructions? 

For context: making an app for quick review of risc-v binaries, looking for an API to retrieve instruction description, seems like this project is a good candidate

ThinkOpenly, 2026-02-10:
What are the expectations here? It's not clear what's being asked, or what the requirements are. If this is still an issue worth keeping open, let's understand what needs to be done to address it. Thanks!

closed 2026-02-10 by lu-zero