riscv/riscv-unified-db #199: index file

drom, 2024-10-25:
Do we have the index yaml file that would allow fetching all the rest of yaml files?

dhower-qc, 2024-10-25:
there is no index; it just assumes every .yaml file under `arch` is included (type is implied by whether it exists in the `inst`, `ext`, etc., subfolder).

You can generate a single file with everything in it with `./do gen:arch` (will appear in `gen/_/arch/arch_def.yaml`

drom, 2024-10-25:
I am fetching it from the web page, and I can't do `ls internet`. I need to know list of the files and folders.

dhower-qc, 2024-10-25:
ah, interesting. we could generate the unified description with each push on main (similar to how we keep https://riscv-software-src.github.io/riscv-unified-db/manual/html/landing/index.html updated). Would that work?

drom, 2024-10-25:
There is no problem having 1k of yaml files, but if we can just regenerate index.yaml file and add it to the same commit. It would help. If it is uses JSON references then it would simplify things:

```json5
{
  arcs: {
    inst: {
      A: {
        'amoadd.d': {'$ref': '#/arcs/inst/A/amoadd.d.yaml'},
        ...

      }
    }
  }
}
```

drom, 2024-10-25:
I can make quick PoC in JS

drom, 2024-10-25:
#201

drom, 2024-10-28:
with index.json I was able to fetch UnifiedDB from GitHub using web client. Here is example:
https://observablehq.com/@drom/risc-v-unifieddb