riscv/riscv-unified-db #203: Common field reference table

drom, 2024-10-28:
Currently, the named fields are belong to each instruction and don't carry instruction based semantics.
Here is example of `addi` instruction with `rs1, rd, imm`

```yaml
addi:
  long_name: Add immediate
  description: Add an immediate to the value in rs1, and store the result in rd
  definedBy: I
  assembly: xd, xs1, imm
  encoding:
    match: -----------------000-----0010011
    variables:
    - name: imm
      location: 31-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
```

Does it make sense to specify reference from each instruction into common reference table that will carry more detailed information about the field. Here is how the table could look like:

https://github.com/drom/riscv/blob/master/lib/fieldo.js

dhower-qc, 2024-10-29:
That's an interesting suggestion, and it would be great to capture more semantic information without having to parse/analyze the IDL code. I'm 100% on board with getting this information in, but just want to think through the best approach.

Generally, we've tried to keep .yaml files under `arch` bare-bones to minimize the dependencies needed to consume the data in any programming language ('flat' as you say). That's why we've avoided things like YAML anchors/refs or, in this case, JSON references.

**So, what to do?**

1. Introduce JSON Reference in .yaml files under arch. We can provide a flat version as a generated artifact with each commit (we should really be doing this anyway).
2. Use JSON Reference in template files that generate .yaml files under arch. We already do something like this in limited cases where there is a lot of repetition (e.g., the pmpaddrN registers, which are generated from pmpadddrN.layout). 
3. Don't introduce JSON Reference, and just add the 'flat' info to the existing .yaml files. We could programmatically detect common fields, but it wouldn't be explicit in the data anywhere.

I'm inclined to go the route of option 1, but just worry that it could be another obstacle making UDB adoption less likely. Especially if JSON Reference becomes a requirement for anyone contributing to the DB (e.g., extension authors).

This is a good topic to discuss in a sync-up meeting. In the meantime, I'm curious what your thoughts are.

** Side note **

This data is related to instruction formats (I-type, R-type, etc.), which has come up a few times since it's not currently captured by the database. This was initially intentional -- in my corner of the world, we generally don't care about them, especially since it gets ugly quick with things like the C extension. However, compiler folks like @apazos have been educating me on why it's important to capture it, _e.g._, for linker relaxation information. It seems like this would be right place to figure out how to integrate the format information.

drom, 2024-10-29:
You already have some implicit references from one file to another, like `definedBy: I`. Implicit links require custom sideband knowledge about the ways to reason components together.
Explicit JSON/YAML references is an explicit composition mechanism that is still "human readable/writable" but allows declarative, low-overhead, language agnostic automatic resolution.
IMHO we should use `$ref`

drom, 2024-10-29:
> This data is related to instruction formats (I-type, R-type, etc.), which has come up a few times since it's not currently captured by the database.

Yes, canonical '*-type' format is an very high level abstraction that is already broken in many places. I don't see much value caring it as is through the machine readable spec. On other hand, the unique set of field types constitutes specific type format. for example:

```yaml
variables:
    - {$ref: 'fieldo.json/imm12'}
    - {$ref: 'fieldo.json/rs1'}
    - {$ref: 'fieldo.json/rd'}
```

We can call it `rd-rs1-imm12-type` format.

AFOliveira, 2024-10-30:
As a follow up on earlier discussion, I'll take the first step towards this and turn the variables name into their own nicknames, if no one has anything against it. After those nicknames are correct, I guess moving to a json schema pass would be just find/replace, correct?

@drom @dhower-qc  Do you think that we should keep values like `imm12hi, imm12lo` separated on the variable list? It seems quite odd because no matter how the encoding is done, it actually is only one variable.

dhower-qc, 2024-10-31:
> As a follow up on earlier discussion, I'll take the first step towards this and turn the variables name into their own nicknames, if no one has anything against it.

Excellent

> After those nicknames are correct, I guess moving to a json schema pass would be just find/replace, correct?

Take a look at #228. I've updated the schema to handle $mref from an instruction variable. (this is different than $ref, which according to JSON Reference should ignore any other fields, which isn't what we want here). I kept the old schema around, too, so we don't have to convert everything now. 

> 
> @drom @dhower-qc Do you think that we should keep values like `imm12hi, imm12lo` separated on the variable list? It seems quite odd because no matter how the encoding is done, it actually is only one variable.

I would prefer we merge fields like that into one so that we maintain a 1:1 mapping between the variable names and the type information.

AFOliveira, 2024-11-02:
> I would prefer we merge fields like that into one so that we maintain a 1:1 mapping between the variable names and the type information

I agree with your view here, problem is that riscv-opcodes has fields like `imm12` and `imm12hi, imm12lo` which mean different fields. 
I think we have at least two approaches to move forward on this:

- We define new nicknames that allow for all the previous encodings, this has the upside of being kind of easy due to being a direct renaming from riscv-opcodes
- We define new nicknames, but ally them to Instruction type, i.e. U-Type, B-Type and use this encondings types to ensure that we never miss any field. This has the upside of using ratified and generic description that anyone would easily understand when reading and writing, and would allow for yet again the speck checking upon itself, due to each instruction type having specific fields.

I am more inclined to the second one, because I think it is more sustainable in the future, but would like to hear your takes on this.

dhower-qc, 2024-11-02:
I agree; the second approach seems better long-term.

ThinkOpenly, 2024-11-11:
FWIW, I'm not a big fan of associating too much meaning to the names of the assembly operands. Looking at what riscv-opcodes has shows that it gets messy. The operand names aren't architecturally significant. You could have an operand called "bimm12hi", or you could call it "X". The important information is the operand order, what the operand represents, and how it is encoded. The order is already present in the "assembly" field, the representation could be included in the "flat" YAML (GPR register file index, although the "types" would then need to be defined somewhere), and the encoding is already present.

(In some ways, it would be nice to use more semantically meaningful names for the operands, like "offset", "addend1" and "addend2", "numerator" and "denominator", but perhaps that ship has sailed.)

AFOliveira, 2024-11-11:
I agreed and defended exactly your point on a previous meeting, however @drom's point to which I also agreed was that 

> important information is the operand order, what the operand represents, and how it is encoded

was going to get duplicated if we didn't nickname. So what we're trying to do is nickname but associate that nickname to a data structure that has all the parameters that actually define that operand.

ThinkOpenly, 2024-11-12:
> > important information is the operand order, what the operand represents, and how it is encoded
> 
> was going to get duplicated if we didn't nickname. So what we're trying to do is nickname but associate that nickname to a data structure that has all the parameters that actually define that operand.

Understood, but I'm more in favor of establishing "types" for the operands, rather than depending on nicknames. While similar, there is a distinction: one issue, IMHO, with nicknames is that it tries to encapsulate both the type and the encoding, as in the example in your comment https://github.com/riscv-software-src/riscv-unified-db/issues/203#issuecomment-2453102503:
> problem is that riscv-opcodes has fields like imm12 and imm12hi, imm12lo which mean different fields.

I propose avoiding the use of riscv-opcodes as a model. Let the operands stand on their own, and let the encoding either be specified (duplicated) with the instruction, or we wade into formats.

We are going to need to include instruction formats at some point. Downstream projects need that information, like binutils ("INSN_CLASS"):
```
/* All RISC-V instructions belong to at least one of these classes.  */
enum riscv_insn_class
{
  INSN_CLASS_NONE,

  INSN_CLASS_I,
  INSN_CLASS_C,
  INSN_CLASS_M,
  INSN_CLASS_F,
  INSN_CLASS_D,
  INSN_CLASS_Q,
  INSN_CLASS_F_AND_C,
  INSN_CLASS_D_AND_C,
```
```
{"addi",        0, INSN_CLASS_I, "d,s,j",     MATCH_ADDI, MASK_ADDI, match_opcode, 0 },
```
QEMU:
```
# Formats 32:
@r       .......   ..... ..... ... ..... ....... &r                %rs2 %rs1 %rd
@i       ............    ..... ... ..... ....... &i      imm=%imm_i     %rs1 %rd
@b       .......   ..... ..... ... ..... ....... &b      imm=%imm_b %rs2 %rs1
@s       .......   ..... ..... ... ..... ....... &s      imm=%imm_s %rs2 %rs1
@u       ....................      ..... ....... &u      imm=%imm_u          %rd
@j       ....................      ..... ....... &j      imm=%imm_j          %rd
```

dhower-qc, 2024-11-12:
Check out https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/inst/I/addi.yaml

That's an example of using the recently added $inherits feature. It should allow us to 'inherit' operand types while keeping the name independent. I agree with Paul that we want an independent name to, for example, distinguish imm vs offset.

AFOliveira, 2024-11-12:
> That's an example of using the recently added $inherits feature

My question might seem dumb, but isn't that just shifiting the nicknames to the end of the inherit?
      `$inherits: ../../common/inst_variable_types.yaml#/**itype_imm**
`

ThinkOpenly, 2024-11-12:
> > That's an example of using the recently added $inherits feature
> 
> My question might seem dumb, but isn't that just shifting the nicknames to the end of the inherit?

I had the same impression. Given that the "location" is pretty closely tied to the instruction format, is it possible to inherit _all_ of the operands and their locations based on the instruction format? (Showing my JSON ignorance here.) Something like:
```
  encoding:
    match: -----------------000-----0010011
    variables:
      $inherits: ../../common/inst_formats.yaml#/itype
```
and the "itype" stanza in `inst_formats.yaml` would have all of the variable fields:
```
  itype:
    - name: imm
      location: 31-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
```

AFOliveira, 2024-11-12:
@ThinkOpenly That approach is great in terms of adding "hints" to each field, something like what happens here in the ISA manual:

[image]

However, some questions arise such as: How are we going to specify imm vs offset in JALR? It uses I-Type, but its' immediate is an offset.

[image]

dhower-qc, 2024-11-12:
> isn't that just shifting the nicknames to the end of the inherit?

I'd say yes and no. Yes, this is giving a nickname to fields. But no, it's not _just_ a nickname. Since it is inheritance, you can overwrite anything you want. And you can decide if you want to use the nicknames or just suck up the flattened view.

> is it possible to inherit all of the operands and their locations based on the instruction format

Maybe. $inherits is really set up to work with hashes (not arrays) so that you can easily identify what you want to overwrite. If you don't want to ever overwrite anything, $ref from JSON Reference (also supported now) works with any data structure but can't be modified. If we want to support inheriting the whole variable encoding _and_ overwriting some fields, then we'll want to restructure the schema so that `variables:` is a hash. If we do that, the $inherits feature will work fine.

Examples for each scenario:

```yaml
path:
  to:
    array_type:
      - name: imm
        location: 31-20
      - name: rs1
        location: 19-15
      - name: rd
        location: 11-7

path:
  to:
    hash_type:
      imm:
        name: imm
        location: 31-20
      rs1:
        name: rs1
        location: 19-15
      rd:
        name: rd
        location: 11-7
```

```yaml
variables:
  $ref: "#/path/to/array_type"

# result is
#
# variables:
#   - name: imm
#     location: 31-20
#   - name: rs1
#     location: 19-15
#   - name: rd
#     location: 11-7
``` 

```yaml
variables:
  $inherits: "#/path/to/hash_type"
  imm:
    name: offset
    
# result is
#
# variables:
#   imm:
#     name: offset
#     location: 31-20
#   rs1:
#     name: rs1
#     location: 19-15
#   rd:
#     name: rd
#     location: 11-7
```

drom, 2024-11-12:
What are we trying accomplish with `$inherits` feature? Just write less of yaml?
Nicknaming fields, or immutable references `$ref` allow reference counting.
Strict set of specific references gives clear type signature to instruction.
Two instructions with exacly-the-same type signature have exactly-the-same type.
With `$inherits` feature, it will be hard to reason exactness of instruction types, despite "similar" set of fields.

dhower-qc, 2024-11-12:
> What are we trying accomplish with $inherits feature? 

I see several benefits:

- Write less YAML (as you mention). It's also easier to maintain (ignoring the $ref alternative for a minute) since updates to the base propagate.
- Add ability to change shared state. The example discussed above shows why you want to - you can have two I-type instructions, but one wants to name a field `imm` and the other `offset`. You can still add a `kind` field in the base that both would inherit to make exactly-the-same type comparisons straightforward.
- The most obviously place for `$inherits` is with the profile specs since releases build on each other. You can now see how that's being used in the database:

https://github.com/riscv-software-src/riscv-unified-db/blob/6287ff91090dfdadb48baa46aa6675ec4b363da4/arch/profile_release/RVA22.yaml#L45-L49

dhower-qc, 2024-11-14:
Was there any resolution for this after I left the call yesterday?

As I was thinking about it more, I remembered that the variable names are used by IDL (in the `operation()` function). We'll need to take that into account.

AFOliveira, 2024-11-15:
@dhower-qc would a ref with an ending in something along the lines of fieldo (https://github.com/drom/riscv/blob/master/lib/fieldo.js) work for the IDL?

Imm would never be defined as two split parameters though

dhower-qc, 2024-11-15:
First, let's list all the requirements we have collected ('field' == 'variable'):

1. Fields, even those split in the encoding, are a single object.
    1. So that they can be mapped in documentation/IDL
2. All fields, even those that are a constant, should be captured
    1. We aren't currently doing this. But there are request to, e.g., capture that `encoding[6:0] == 0010011` means 'OP-IMM'. 
4. Field keys should be decoupled from a display name (e.g., many fields with a display name "imm" will map to different data.). Two reasons for this:
    1. imm12pc probably isn't what we want to show in documentation/disassembly.
    3. Similarly, we don't want to have to use the key names in IDL
5. Instructions are associated with an ABI instruction type (e.g., I-Type).
6. Fields contain:
    1. location in the instruction
    2. implicit shift, if any
    4. data type (dest reg, src reg, signed imm, unsigned imm, flag)
    5. riscv-opcode name(s)
    7. value, if constant
    8. restrictions (NOT, LESS THAN, etc.), if any
9. Instructions provide an assembly format using field "display names"

Now, back to your question (can we use fieldo format?):

If it works, yes! As I tried to articulate in the meeting, I'm just worried we are creating a maintenance problem by trying to capturing all the requirements above in an immutable `$ref`. Thinking through a potential format for a few instructions:

```yaml
kind: instruction
name: addi
abi_type:
  $ref: common/inst_abi_types.yaml#/$defs/itype # what actually goes in here?
assembly: rd, rs1, imm
encoding:
  mask: # ... this is partially redundant if we capture named const fields, too
  variables:
  - $ref: common/inst_variable_types.yaml#/$defs/simm12_display_name_imm
  - $ref: common/inst_variable_types.yaml#/$defs/xd_display_name_rd
  - $ref: common/inst_variable_types.yaml#/$defs/xs1_display_name_rs1
  - $ref: common/inst_variable_types.yaml#/$defs/enc32_op_imm_major_opcode
  - $ref: common/inst_variable_types.yaml#/$defs/enc32_funct3_addi_opcode
---
kind: instruction
name: lb
abi_type:
  $ref: common/inst_abi_types.yaml#/$defs/itype # what actually goes in here?
assembly: rd, offset(rs1)
encoding:
  mask: # ...
  variables:
  - $ref: common/inst_variable_types.yaml#/$defs/simm12_display_name_offset
  - $ref: common/inst_variable_types.yaml#/$defs/xd_display_name_rd
  - $ref: common/inst_variable_types.yaml#/$defs/xs1_display_name_rs1
  - $ref: common/inst_variable_types.yaml#/$defs/enc32_load_major_opcode
  - $ref: common/inst_variable_types.yaml#/$defs/enc32_funct3_lb_opcode
```

There are few red flags to me:

1. Every instruction needs at least one unique field key (identifying its minor opcode -- enc32_funct3_lb_opcode).
2. Small changes require a new key (display name imm -> offset)

Maybe I'm just not being imaginative enough.

AFOliveira, 2024-11-15:
Well, I definitely was not expecting this task to turn itself so challenging, but I'll try my best to keep up :)

First of all, I mostly agree with your concerns, I'm just trying to create a prototype so that everyone can critique and we get a better sense on how to make this work.

My considerations on the requirements are the following:

1. Fully agree
2.  I see the need for this, it would even enhance documentation, I have even been able of extracting this info from riscv-opcodes in the past, so I think I can help with that. IMO  In the DB part, this could be done by setting the instruction type as the most important parameter and then continuing by assigning each part of those encodings, right? I see two approaches here, we could do as you just examplified and let the back-end assign each parameter to each ref,  or we could start those assignments in the yaml, i.e.  variable fields $ref would not need to set encoding bit position, since this would belong to the I-type field(check the enf of the issue for context). What of this approaches do you think would be better?
3.  What would this nickname be? Just immSIZE, e.g. imm12? How would people now each variable characteristics? A variable table could possibly solve this.
4. This refers back to 2.
5. Agree with all but with iv. Why would we need this? It is not much work at all to do it, but I just don't comprehend any possible advantage that it could generate.
6. Shouldn't this be a _vii_ point in 5. then?

**Aditional context for 2.:**
```
  variables:
  - imm12 =  $ref: common/inst_variable_types.yaml#/$defs/simm12_display_name_imm
  - rd = $ref: common/inst_variable_types.yaml#/$defs/xd_display_name_rd
  - rs1 = $ref: common/inst_variable_types.yaml#/$defs/xs1_display_name_rs1
  -  opcode = 0x10
  -  funct3 = 101

```