riscv/riscv-unified-db #206: bit field gathering

drom, 2024-10-29:
Not all logical fields (variables) inside instruction have consecutive location ranges like in example below:

```yaml
addi:
  long_name: Add immediate
  description: Add an immediate to the value in rs1, and store the result in rd
  definedBy: I
  assembly: xd, xs1, imm
  encoding:
    match: -----------------000-----0010011
    variables:
    - name: imm
      location: 31-20
    - name: rs1
      location: 19-15
    - name: rd
      location: 11-7
```

We should provide more universal declarative mechanism to gather bits from instruction body into each field.

The good example is 'C' extension:

[image]

https://observablehq.com/@drom/risc-v-c-opcodes

Green fields are holding some bits for the immediate literals in very convoluted order.

It would make sense to specify array of indexes where bits are taken from in desired order.

Example formats for `C` extension fields:

https://github.com/drom/riscv/blob/master/lib/fieldo.js#L63

dhower-qc, 2024-10-29:
This is already a feature. The C extension isn't pushed publicly yet, but @ayosher in Qualcomm is finishing it up. It handles the non-consecutive bits and things like a variable "!= 0".

[jal](https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/inst/I/jal.yaml) is a good example of how more complex encodings are handled:

```yaml
  encoding:
    match: -------------------------1101111
    variables:
    - name: imm
      location: 31|19-12|20|30-21
      left_shift: 1
      sign_extend: true
    - name: rd
      location: 11-7
```

After that gets parsed, the generated encoding diagram is:

[image]

and the variable, in IDL are:

```
signed Bits<21> imm = sext({$encoding[31], $encoding[19:12], $encoding[20], $encoding[30:21], 1'd0});
Bits<5> rd = $encoding[11:7];
```

drom, 2024-10-29:
Sounds good!

closed 2024-10-29 by drom