riscv/riscv-unified-db #208: assembly syntax

drom, 2024-10-29:
I am looking at `assembly` field for instruction and wondering. Should it be in the order/syntax/optionallity of actual assembly language?

Here is an example from here https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/inst/V/vfadd.vf.yaml#L8

```yaml
vfadd.vf:
  long_name: No synopsis available.
  description: |
      No description available.
  definedBy: V
  assembly: vm, vs2, xs1, vd
  ...
```

An actual assembly order is:

```asm
vfadd.vf   vd, xs1, vs2[, vm]

vm - is optional
```

or:

```yaml
vlse8.v:
  long_name: No synopsis available.
  description: |
      No description available.
  definedBy: V
  assembly: nf, vm, xs2, xs1, vd
  ...
```

in assembly:

```asm
vlse8.v  vd, (xs1), xs2[, vm][, nf]
```

or:

```yaml
bne:
  long_name: Branch if not equal
  description: |
    Branch to PC + imm if
    the value in register rs1 is not equal to the value in register rs2.

    Raise a `MisalignedAddress` exception if PC + imm is misaligned.
  definedBy: I
  assembly: xs1, xs2, imm
   ...
```

```asm
bne  xs1, xs2, <pc + imm>
```

We clearly need a better way to express these details.

dhower-qc, 2024-10-30:
Yep. We definitely need a better way. It would be good to get a list of all the odd cases so we can think through a format.

@AFOliveira, where did you pull the vector assembly format from?

drom, 2024-10-30:
yes, riscv-opcodes don't provide correct order or syntax for the assembly language. We need to reference to the adoc and test it with gcc / clang to get it right

AFOliveira, 2024-10-30:
I'll try to come up to a solution for the assembly generation. I think it may be possible to parse it from the toolchain in some way. I'll follow up here whenever I find have any updates.

drom, 2024-11-01:
RVV is quite complex. here is repo with intrinsics :
https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/main/archive/rvv-intrinsic-api.md

AFOliveira, 2024-11-01:
Thank you. I'll look into it aswell.

dhower-qc, 2024-11-02:
Here is how binutils represents assembly formats for RV:

https://github.com/bminor/binutils-gdb/blob/master/opcodes/riscv-opc.c

You can see how the string maps in the file riscv-dis.c in the same directory. 

We don't necessarily have to follow the same format, but it is at least a solid reference.

AFOliveira, 2024-11-02:
I was already thinking of using binutils for this task, I think it will also be helpful in getting instructions types as mentioned in #203. For now, I will keep the my tool getting encodings from riscv-opcodes, but if we can replace it, I think I can scrape all the instruction data from binutils.

ThinkOpenly, 2024-11-11:
We already pull the operands directly out of Sail, although I'm not positive we get everything correct currently.
```
    {
      "mnemonic": "vfadd.vv",
      "name": "TBD",
      "operands": [
        {
          "name": "vd",
          "type": "regidx",
          "optional": false
        },
        {
          "name": "vs2",
          "type": "regidx",
          "optional": false
        },
        {
          "name": "vs1",
          "type": "regidx",
          "optional": false
        },
        {
          "name": "vm",
          "type": "bits(1)",
          "optional": true,
          "default": "v0.t"
        }
      ],
```
Would this be helpful?

drom, 2024-11-12:
> We already pull the operands directly out of Sail, although I'm not positive we get everything correct currently.
>
> ```
>     {
>       "mnemonic": "vfadd.vv",
>       "name": "TBD",
>       "operands": [...]

That is half way to where we want to be. `"type": "regidx"` is no quite enough. We need more info about the field.

* which register file do we reference: int, float, vector, ... ?
* is it the source, destination or both?
* is it zero/sign extended?
* formula: pc + (imm12 << 1)
* etc.

Here are examples:
https://github.com/drom/riscv/blob/master/lib/fieldo.js#L29