riscv/riscv-unified-db #210: Implement V in IDL

dhower-qc, 2024-10-30:
The entire Vector (V) extension needs an IDL implementation.

AFOliveira, 2025-03-18:
If I understand correctly this is currently being done in #467 by @jmawet .  I added the correct tags so we can just mange all active development better. If I am wrong, please correct me

ThinkOpenly, 2025-03-18:
It's a really big task, is it something that could be a candidate for "divide and conquer"? We have 2 IDL Writer mentees that might be helpful.

jmawet, 2025-03-18:
Yeah I've been spending a lot of my free time reading through the spec trying to understand the vector instructions. I definitely get the purpose of this udb project, the spec is so confusing haha, especially trying to figure out each individual instruction. My progress is moving at a snail's pace I feel. 

Combine that with the huge task of creating a vector implementation in IDL... maybe I've bitten off more than I can chew. If we have IDL writer mentees, they might be better suited to this, especially implementing idl vector functions

latifbhatti, 2025-03-25:
Hi @jmawet ,
First of all, I really appreciate the work you’re doing on the RISC-V vector extension for the IDL UDB project! It’s great to see this being developed, and I’m excited to contribute.

I wanted to touch base regarding the RISC-V vector extension work in the IDL UDB project. Abdullah and I have been selected as mentees for this project, and I’d like to ensure that our contributions align with the existing implementation.

I have experience with the RISC-V vector extension and have developed an open-source RISC-V vector co-processor (Vaquita). 

However, I’m not sure how much progress has already been made on the vector extension in IDL. To avoid duplicate work and ensure an efficient divide and conquer approach, I’d love to discuss:

    The current status of the vector extension implementation in IDL

    Areas where contributions are needed

    How we can best divide tasks to accelerate progress

Would it be possible to have a quick sync to go over these points? Alternatively, if there are any documents or ongoing discussions I should review, please point me in the right direction.

Looking forward to collaborating!

latifbhatti, 2025-03-25:
@dhower-qc , @ThinkOpenly and @jmawet 

I appreciate the ongoing development of the RISC-V vector extension for IDL UDB! Abdullah and I have been selected as mentees for this project, and we want to ensure our contributions align with the existing implementation. Could you share the current status, areas that need contributions, and how we can effectively divide tasks? Also, what’s the best way to get started with the IDL project? Looking forward to collaborating!

jmawet, 2025-03-25:
Hi! 
IDL info should be here, but it's giving me 404 right now: https://riscv-software-src.github.io/riscv-unified-db/example_cfg/html/generic_rv64/prose/idl.html

I'll talk to @dhower-qc about setting up a meeting with all of us to discuss. Sounds like you've got good experience with RVV/core arch (especially compared to me, I work in physical design mostly so this project is a big learning experience for me, I'm just contributing for funzies when I am able) I'm thinking y'all will probably take over the bulk of the vector extension/idl effort, and I can help out filling in individual instructions based on what y'all have come up with.

dhower-qc, 2025-03-25:
> IDL info should be here, but it's giving me 404 right now: https://riscv-software-src.github.io/riscv-unified-db/example_cfg/html/generic_rv64/prose/idl.html

It moved here: https://riscv-software-src.github.io/riscv-unified-db/example_cfg/html/example_rv64_with_overlay/prose/idl.html

latifbhatti, 2025-03-25:
Hi @jmawet ,
That sounds like a great plan! I appreciate you reaching out to set up the meeting. It'll be helpful to get everyone aligned on the vector extension and IDL effort.

Your experience in physical design is definitely valuable, and it's great to have your contributions whenever you're available. We’ll take the lead on the broader vector extension/IDL implementation, and your help with filling in individual instructions will be really useful.

Looking forward to discussing the remaining work for the vector extension in the IDL project during the meeting!

ThinkOpenly, 2025-03-25:
> Could you share the current status, areas that need contributions, and how we can effectively divide tasks? Also, what’s the best way to get started with the IDL project? Looking forward to collaborating!

A meeting might be a good idea. If you want to dive in, I'd suggest picking a reasonably simple vector instruction, like "Vector Add". There are 3 variants:
- vector+vector
- vector+scalar
- vector+immediate

None are particularly more difficult than others. The "difficult" parts include accounting for VLEN, SEW, LMUL, vta and vma masks, vl vector length, vstart, etc. as applicable.

ThinkOpenly, 2026-08-20:
@jpak-qc, FYA: note that we'll eventually need to accommodate Zvfbfa's requirement that when `vtype.ALTFMT` is set, all vector floating-point instructions become reserved, except for the set that accommodates BF16. We can open a new issue for that if it would help to keep it from being lost.