riscv/riscv-unified-db #213: Add RVA23 Profile

dhower-qc, 2024-10-30:
**Describe the data**

Need to add the RVA23 profile to the database. Will also require minor changes to the database schema to capture new types of optional in RVA23.

james-ball-qualcomm, 2024-10-30:
I've started working on #58 to reduce the duplicate Ruby and Embedded Ruby code between CRDs and Profiles. This is likely to require some CRD/Profile schema changes in their YAML files. So, my request is to wait until that is complete (likely next week) before we add RVA23/RVB23 profiles.

james-ball-qualcomm, 2024-11-14:
Just realized RVB23/RVA23 might reference some new extensions/instructions/CSRs not in the UDB yet. Let us know what you find.

ThinkOpenly, 2025-03-07:
> I’ve already added those missing extensions and more for RVA23 but in my fork of UDB. How should we handle this? I want to avoid a merge conflict down the road.

What are your plans for getting those changes merged? Are they visible on GitHub?