riscv/riscv-unified-db #215: Add Sdext (External debug arch)

dhower-qc, 2024-10-30:
**Describe the data**

Need to add Sdext to the database. Also need IDL with "builtin" interface to interact with an external debugger model.

ThinkOpenly, 2025-07-16:
Some of this is done (PR #350), some of this is pending merge (#614). Some is going to be pretty challenging:

> Also need IDL with "builtin" interface to interact with an external debugger model.

Not saying "no", but there might be lower-hanging fruit which would be a better "good first issue". (And we might want to remove that tag from this issue, frankly.)

ThinkOpenly, 2025-07-16:
> Are there any easier tasks or ‘good first issues’ you recommend for someone new to the project?

Perhaps #223?

ThinkOpenly, 2025-10-26:
The work here should also include defining the `dret` instruction.

Goyamjain06, 2026-10-04:
I'd like to take on the remaining data work for this issue: filling in the dret
instruction (currently a placeholder), writing a complete Sdext description,
and adding any missing parameters. Per the discussion above, I'll scope this
PR to the data layer only and leave the IDL debugger-interface builtins for a
follow-up, since that part was flagged as more involved.

@shivaansh0610-LUFFY — if you're already working on this, happy to help with
review instead or pick up something else. Otherwise I'll get started on the
above.

Goyamjain06, 2026-10-04:
Did some reading of the external debug spec (riscv/riscv-debug-spec) before
drafting. Two things worth confirming before I write the YAML:

1. `dret` outside Debug Mode — the spec doesn't define this behavior (the only
mention is in a non-normative appendix). Should `access` be `never` for
s/u/vs/vu, i.e. illegal instruction outside Debug Mode, or is there a reason
it's currently `always`?

2. Possible existing bug: dcsr.yaml's four parameters (DCSR_STEPIE_TYPE,
DCSR_STOPCOUNT_TYPE, DCSR_STOPTIME_TYPE, DCSR_MPRVEN_TYPE) are marked
`definedBy: Sdtrig`, but dcsr itself belongs to Sdext, not Sdtrig. Should
these move to Sdext as part of this PR, or is that a separate fix?

Happy to proceed with my best judgment on both if there's no response in a
day or two, just flagging before I commit to an approach.

Goyamjain06, 2026-10-07:
Opened a PR covering the data-layer scope discussed above: #2725

It fills in the Sdext extension description and the dret instruction
(operation(), access, privilege-mode restore from dcsr.PRV/V, MPRV/MDT
clearing, dcsr.PELP clearing under Zicfilp). Left as follow-up work:
the IDL debugger-interface builtins, the Ssdbltrp/Zicfilp TODOs (UDB
doesn't model those fields/state yet), and the dcsr.yaml parameter
definedBy:Sdtrig fix from earlier — happy to take those on separately
once this is reviewed.