riscv/riscv-unified-db #217: Add Smclic

dhower-qc, 2024-10-30:
**Describe the data**

Add extension (currently in development state).

Add hart-visible additions in IDL where necessary.

ThinkOpenly, 2025-07-16:
> pls assign me would make pr for it

Since Smclic is still unratified, and I just assigned #410 to you, let's hold on assigning this one for the moment.

AFOliveira, 2025-07-16:
Great to see your interest @blazethunderstorm!

Sukuna0007Abhi, 2025-07-18:
Well hello @ThinkOpenly @AFOliveira can I work on it sir?

AFOliveira, 2025-07-21:
@Sukuna0007Abhi Let's go one PR at a time, how does that seem?

Sukuna0007Abhi, 2025-07-21:
Ok sir but I will always have a eye on this also. Thanks @AFOliveira

ReVuz, 2026-10-21:
Hi @dhower-qc and @ThinkOpenly!  I'd be interested in contributing to this issue. I noticed that Smclic was previously marked as awaiting ratification. Is this issue still deferred, or would it be okay to start working on the YAML definitions and any necessary IDL additions? I'd be happy to help if it's ready to be picked up.

ThinkOpenly, 2026-10-21:
A lot has changed here. The family of extensions is now known as ACLIC (Advanced Core-Local Interrupt Controller), and the latest released specification is version 0.2. They call this "stable", although I would expect a much higher release number (like 0.7).

There's no harm in starting to define the extensions here, if you are willing and able.

The specification can be found at https://github.com/riscv/riscv-fast-interrupt.