riscv/riscv-unified-db #219: Add Smepmp

dhower-qc, 2024-10-30:
**Describe the data**

Add Smepmp extension

MuhammadHammad001, 2025-04-09:
Hi @dhower-qc If this is not assigned to anyone or nobody has started to work on it, Can I please work on this as I have written few tests for this extension in my company and I have an idea about the PMP as well?

ThinkOpenly, 2025-04-09:
Assigned to you, @MuhammadHammad001. Thanks!

We've often broken the extensions into pieces:
- the extension definition in `arch/ext`
- the extension-defined CSRs in `arch/csr/[extension]`
- the extension-defined instructions in `arch/inst/[extension]`

So, feel free to break up this task as needed.

MuhammadHammad001, 2025-04-09:
Thank you @ThinkOpenly .

dhower-qc, 2025-09-03:
Needs assignee to pick up #652

closed 2026-01-29 by dhower-qc