riscv/riscv-unified-db #222: Add Zacas (Atomic CAS)

dhower-qc, 2024-10-30:
**Describe the data**

Add spec + IDL

syedowaisalishah, 2025-03-18:
@dhower-qc and @ThinkOpenly , I am interested in working on this issue. Could I take it up? If so, could you provide any guidance on how to proceed?

dhower-qc, 2025-03-18:
For the IDL, this is relatively straightforward. IDL treats atomic memory operations as a function call to the SoC model, so what needs written in IDL is everything around the physical memory access. Take a look at the atomic memory operations from Zaamo (like amoadd.w) to get a more concrete idea.

syedowaisalishah, 2025-03-18:
Thank you for assigning me this issue!

I will start by adding synopsis and description for the three instructions using the ISA spec. Could you provide a link to the ISA spec for Zacas, or let me know where I can find it?

For IDL, I will refer to Zaamo's atomic operations (amoadd.w) to understand the structure. Should I follow the same format as Zaamo for writing the IDL? Also, where should I add the new IDL definitions?

ThinkOpenly, 2025-03-18:
> Could you provide a link to the ISA spec for Zacas, or let me know where I can find it?

riscv.org, under "Specifications" (also posted in Slack)

> For IDL, I will refer to Zaamo's atomic operations (amoadd.w) to understand the structure. Should I follow the same format as Zaamo for writing the IDL? Also, where should I add the new IDL definitions?

Same format as A/Zaamo, so the IDL goes with the "operation()" field in the corresponding YAML.

jordancarlin, 2025-03-18:
> The hard steps are to write the IDL. (The Sail needs to be written, too, be we've leaned entirely on the sail-riscv project to provide that.)

Just in case this proves useful, there is an open PR with the Sail for Zacas: https://github.com/riscv/sail-riscv/pull/285. Needs a few updates, but should be fairly close to what the final version ends up looking like.

syedowaisalishah, 2025-03-18:
Thanks for the guidance!

dhower-qc, 2025-03-19:
replacing with #516, #517

closed 2025-03-19 by dhower-qc