riscv/riscv-unified-db #223: Add Zabha (Byte and Halfword Atomics)

dhower-qc, 2024-10-30:
**Describe the data**

Add isa spec and IDL

ThinkOpenly, 2025-07-16:
FYI, there are some references for atomic memory operations in `spec/std/isa/inst/Zaamo`.

Another aspect of this work is that all of the "AMO" instructions have 4 variants. For example, `amoadd.h` has [ `amoadd.h`, `amoadd.h.aq`, `amoadd.h.rl`, `amoadd.aq.rl` ]. This can be handled by replicating everything, or (preferably) by using the "layout" functionality that is used for some CSR sets. Examples of layout use are in `Rakefile` and by looking for `*.layout` files.

Sukuna0007Abhi, 2025-07-19:
Hello @ThinkOpenly @dhower-qc  can I work on this issue?

AFOliveira, 2025-07-20:
Hello @Sukuna0007Abhi! I see this is going to be your first UDB contribution, thanks for the interest!
I will assign this to you so you can complement what @ddrcode added, but please note that majority of what is left is IDL code (I think) so it may take some time to get it right. Of course, we are here to help and you can just tag us for help along the way as many times as you can! Ready for this?

Sukuna0007Abhi, 2025-07-21:
Thanks @AFOliveira I assure you that I won't disappoint you.I will solve this problem with my 100% effort.