riscv/riscv-unified-db #245: Create backend to replace Profiles & Bases & Extensions speadsheet

james-ball-qualcomm, 2024-11-06:
Here's a link to the current spreadsheet to mimic the output of: https://docs.google.com/spreadsheets/d/1A40dfm0nnn2-tgKIhdi3UYQ1GBr8iRiV2edFowvgp7E.

I'll create the backend (unless @dhower-qc beats me to it) and work with @dlweaver to ensure the generated equivalent from UDB is consistent with the spreadsheet. How's that sound?

dlweaver, 2024-11-06:
From: james-ball-qualcomm ***@***.***>
Sent: Wednesday, November 6, 2024 4:11 AM

Here's a link to the current spreadsheet to mimic the output of: https://docs.google.com/spreadsheets/d/1A40dfm0nnn2-tgKIhdi3UYQ1GBr8iRiV2edFowvgp7E.

I'll create the backend (unless @dhower-qc<https://github.com/dhower-qc> beats me to it) and work with @dlweaver<https://github.com/dlweaver> to ensure the generated equivalent from UDB is consistent with the spreadsheet. How's that sound?
Good plan – “Book it, Danno!”
= Dave

dhower-qc, 2024-11-06:
> I'll create the backend (unless @dhower-qc beats me to it) and work with @dlweaver to ensure the generated equivalent from UDB is consistent with the spreadsheet. How's that sound?

No danger there 😄

james-ball-qualcomm, 2025-02-24:
Found the write_xlsx Gem that looks like just what is needed: https://rubygems.org/gems/write_xlsx/versions/0.83.0

closed 2025-05-20 by james-ball-qualcomm