riscv/riscv-unified-db #260: add automation for consistent formatting and linting

kbroch-rivosinc, 2024-11-13:
Consistently formatted yaml, json, etc makes it easier for multiple developers and reduces git diffs.

Formatting and linting that can be run on at git commit time with a git hook, reduces the need for this later in PRs (Github action can be used to confirms that it was done)

There are tools to manage pre-commit hooks like: 

* https://pre-commit.com
* https://typicode.github.io/husky/how-to.html

Since this project is a multilingual project:

```
❯ tokei -C --sort lines
===============================================================================
 Language            Files        Lines         Code     Comments       Blanks
===============================================================================
 HTML                  136        94823        55057            0        39766
 YAML                 1146        93645        81165         2961         9519
 Ruby                   34        16690        11701         2340         2649
 JSON                   12         5883         5855            0           28
 JavaScript              7         4158         4096            9           53
 Ruby HTML              20         3001         2668            0          333
 Rakefile               10         1914         1582           38          294
 AsciiDoc                7         1816         1397            0          419
 CSS                     3          556          512            7           37
 BASH                    9          219          165           16           38
 Module-Definition       1           37           25            0           12
 Markdown                1            1            0            1            0
===============================================================================
 Total                1386       222743       164223         5372        53148
===============================================================================
```

I would suggest using pre-commit to manage the hooks but I'm open to exploring other tooling. (I do realize that since pre-commit is python based I'd be adding another language to the dev dependencies)

I'm happy to put in a PR once the discussion done.

dhower-qc, 2024-11-13:
Love that idea. As mentioned in [Issue #246](https://github.com/riscv-software-src/riscv-unified-db/issues/246#issuecomment-2460800411), it might also be nice to have the keys sorted in YAML files to ensure scripted changes produce reasonable diffs (Since different YAML libraries will likely emit in different orders).

kbroch-rivosinc, 2024-11-13:
I did some exploration of yaml formatting with following tools:

* [yq](https://github.com/mikefarah/yq)
* [prettier](https://github.com/prettier/prettier)

Here's my findings:

* yq:
  * pros: 
    * does general formatting
    * sorting of keys `yq --inplace  '... | sort_keys(.)'`
  * cons: 
    * formatting replaces multi-line "description" values with quotes and "\n"
    * sorting can't be customized
* prettier:
  * pros:
    * does general formatting (doesn't touch multiline description)
  * cons:
    * no sorting capabilities
    
My recommendation based on this would be to move ahead with prettier formatting and writing a custom script to enforce keyword sorting with a specific ordering for some keyword if needed.

kbroch-rivosinc, 2024-11-20:
Discussed this more with maintainers and going to reduce the number of hooks in this initial PR to just very simple:

* end of line
* end of file
* symlink

The other hooks in this initial can then be put in later PRs.

PR #266 (2024-12-10): initial pre-commit config
soft launch meaning only people with pre-commit tool installed would see this run. (full launch would entail, running pre-commit on all files to clean up past and  github action to enforce pre-commit)

If you want to see full effect on files run: `pre-commit run --all-files` I looked the results of this over and changes seem fine.

There's a lot of formatting changes to the html. (but some of the html looks minified so to be expected) I'd suggest maintainers look over what the changes would be for all the files and the generate final docs.

fixes #260
closed 2024-12-10 by dhower-qc