riscv/riscv-unified-db #267: illegal comment in json file flagged by pre-commit check-json

kbroch-rivosinc, 2024-11-14:
This might be a don't care but currently this file: 
https://github.com/riscv-software-src/riscv-unified-db/blob/main/.devcontainer/devcontainer.json#L12
is marked as `.json` and therefore can't have comments.

```bash
❯ pre-commit run --all-files check-json
check json...............................................................Failed
- hook id: check-json
- exit code: 1

.devcontainer/devcontainer.json: Failed to json decode (Expecting property name enclosed in double quotes: line 12 column 3 (char 359))
```

If this format is acceptable for devcontainers I'd suggest renaming it `.json5` 
If not we can just ignore it when check-json is put in place.

dhower-qc, 2024-11-14:
I'm not sure we can rename it to `.json5`. According to the [devcontainer spec](https://containers.dev/implementors/spec/#devcontainerjson), this file must be named `devcontainer.json`. Further, the spec says it is "JSON with comments," not specifically JSON5.

kbroch-rivosinc, 2024-11-14:
Ok
In the short term, this file can just be removed from `check-json` hook. 

In the long term, I'll see about getting a `check-json-with-comments` in pre-commit-hooks repo where it has already been discussed here:
* https://github.com/pre-commit/pre-commit/issues/1320
* https://github.com/pre-commit/pre-commit-hooks/issues/395

AFOliveira, 2025-03-18:
@kbroch-rivosinc is this fixed?

kbroch-rivosinc, 2025-03-18:
yes, we excluded the devcontainer jsonc file: https://github.com/riscv-software-src/riscv-unified-db/blob/main/.pre-commit-config.yaml#L22

closed 2025-03-18 by kbroch-rivosinc