riscv/riscv-unified-db #280: discuss moving to JSON schema 2020-12 (from draft-07)

kbroch-rivosinc, 2024-11-18:
The latest stable json schema is 2020-12: https://json-schema.org/specification 
whereas draft-07 is 2 versions behind.

Here's release notes:

* https://json-schema.org/draft/2019-09/release-notes
* https://json-schema.org/draft/2020-12/release-notes

dhower-qc, 2024-11-18:
Yea, there are a few things out of 2019-09 in particular that would be nice. I picked draft07 because that is the version supported by the [YAML extension in VSCode](https://marketplace.visualstudio.com/items?itemName=redhat.vscode-yaml) that gives you live feedback.

Is there a feature(s) from 2019/2020 that you are particularly interested in?

kbroch-rivosinc, 2024-11-18:
Nothing specific, just looking at the new features in general and figured starting on the latest stable would be better.
I did figure there might be some downstream SW that we wanted that wasn't compatible.

It looks like there's been some movement for that vscode extension:

* https://github.com/redhat-developer/vscode-yaml/issues/704
* https://github.com/microsoft/vscode-json-languageservice/issues/122

dhower-qc, 2024-11-22:
Unfortunately, it seems that there is no activity in the past six months on the YAML extension. Since it's unlikely there will be a version bump any time soon, I suggest we close this issue. We can circle back if activity picks back up (there are 15M installs, so there is a chance someone adopts maintenance).

kbroch-rivosinc, 2024-11-23:
Agreed.  The newer dialects seem overly complicated which I think could lead to slower support. 

There was an interesting discussion where someone propose that semver be put into a json schema dialect as a "format" (like date, uri, or email).  That was rejected and the suggestion was that "$vocabulary" from the 2019-09 dialect be used but that seems like a more complicated solution than just taking the regex here: https://semver.org/#is-there-a-suggested-regular-expression-regex-to-check-a-semver-string

closed 2024-11-23 by kbroch-rivosinc