riscv/riscv-unified-db #287: Why is the U extension "privileged"? This is probably a typo.

james-ball-qualcomm, 2024-11-19:
[image]

dhower-qc, 2024-11-19:
Not a typo: The U extension introduces U-mode and the CSRs to control it (from M/S mode).

james-ball-qualcomm, 2024-11-19:
Glad I asked.

james-ball-qualcomm, 2024-11-19:
What exactly does privileged mean in the schema for a CSR? I'm re-opening this as a documentation request.

From https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/README.adoc

[image]

dhower-qc, 2024-11-21:
My understanding that "privileged" shows up in the priv manual and "unprivileged" shows up in unpriv. That said, those fields are not used anywhere. I only added them because RVI processes seem to make a distinction (though with the combined ISA manual, maybe it really doesn't matter anymore).

james-ball-qualcomm, 2024-11-22:
This raises a good point. I don't see a "U" extension defined in either the Priv or Unpriv ISA manual. So, is the "U" extension actually something we've invented in the UDB? I think we should add some way to mark extensions like Sm and Smpmp that we've created in UDB.

Here's the best I can find about User-mode. 

[image]

dhower-qc, 2024-11-22:
Yes, that's the only place I know where it is mentioned. So, the intent seems to be that U is an extension, but exactly what that means has to be inferred.