riscv/riscv-unified-db #293: automation for human language linting of prose in yaml files

kbroch-rivosinc, 2024-11-20:
To consistently apply style rules across the spec being written by many people, automation could be used.

One such solution "vale" is already being worked with https://github.com/riscv-admin/docs-sig/issues/5

Just want to discuss the deployment of it here.

It could start small even with solving the problem of inconsistency with sentences ending with punctuation seen here:

```
~/rvi/repos/riscv-software-src/riscv-unified-db/arch/inst/I on main ?8                                                                                    Py 3.12.4 Ruby 3.3.4 at 07:56:53 AM
❯ yq ".name,.description"  add*.yaml
add
Add the value in rs1 to rs2, and store the result in rd.
Any overflow is thrown away.

---
addi
---
Add an immediate to the value in rs1, and store the result in rd
---
addiw
---
Add an immediate to the 32-bit value in rs1, and store the sign extended result in rd
---
addw
---
Add the 32-bit values in rs1 to rs2, and store the sign-extended result in rd.
Any overflow is thrown away.
```

dhower-qc, 2024-11-21:
I'm all for this.

dhower-qc, 2024-12-18:
Here's a thought: if we go forward with parallel "ratified" and "community" prose in the generated docs, we could start applying vale to the community text immediately. I.e., we could start to deploy this now for any prose stored in UDB files.

AFOliveira, 2025-03-18:
Is this something that we now can bring to the Doc SIG and maybe ask for some help?

dhower-qc, 2025-03-18:
yes, FWIW, RVI staff is cleared to work on UDB. Kersten is adding the glossary to start and Bill is going to work with me on the ISA manual integration / antora.

AFOliveira, 2025-03-18:
> yes, FWIW, RVI staff is cleared to work on UDB. Kersten is adding the glossary to start and Bill is going to work with me on the ISA manual integration / antora.

Awesome!

kbroch-rivosinc, 2025-03-18:
Currently vale doesn't support yaml as an input source/format but work is being done on it: https://github.com/errata-ai/vale/issues/769#issuecomment-2574496984

So to work with vale as it stands, we'd want to extract the values from the keys we want and pass them to vale directly.

From the command line that roughly looks something like:

setup:
```
❯ wget https://raw.githubusercontent.com/riscv/docs-dev-guide/442db7b78a75643fa6b2ca6cea4f4eb322da44d3/.vale.ini
--2025-03-18 14:54:24--  https://raw.githubusercontent.com/riscv/docs-dev-guide/442db7b78a75643fa6b2ca6cea4f4eb322da44d3/.vale.ini
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.108.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1015 [text/plain]
Saving to: ‘.vale.ini’

.vale.ini                                    100%[=============================================================================================>]    1015  --.-KB/s    in 0s

2025-03-18 14:54:25 (17.3 MB/s) - ‘.vale.ini’ saved [1015/1015]

❯ vale sync
Syncing config [1/1] █████████████████████████████████████████████████ 100% | 0s
 SUCCESS  Synced 1 package(s) to '/Users/kbroch/rvi/non-repo-repos/riscv-unified-db/styles'.
```

run:
```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main [!?] via  v23.10.0 via 🐍 v3.13.2 via 💎 v3.2.3
❯ yq ".description" arch/*/*/*.yaml | wc -l
    7901

kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main [!?] via  v23.10.0 via 🐍 v3.13.2 via 💎 v3.2.3
❯ yq ".description" arch/*/*/*.yaml | vale
✔ 0 errors, 0 warnings and 0 suggestions in stdin.
```

I don't think the referenced `.vale.ini` file is configured to do anything but install the vocabulary right now but once that's established, this could be automated (in a better way).

kbroch-rivosinc, 2025-04-10:
@wmat : mentioned in UDB meeting yesterday that vale supports https://vale.sh/docs/formats/front-matter
So wondering if we can pass in a full yaml file (as front matter) and target the description in the scope of rules.

wmat, 2025-04-10:
Sure can, see the release notes for v3.11.0 here:
https://github.com/errata-ai/vale/releases