riscv/riscv-unified-db #296: discuss refactor cert_model_schema.json and mc100.yaml: extensions should be array of extensions;  parameters should be an array of parameters

kbroch-rivosinc, 2024-11-21:
Similar to the change for inst_schema.json, I don't think we want JSON  keywords of actual extension names or of actual parameters.

example: mc100.yaml
```yaml
extensions:
  I:
    version: "~> 2.1"
    presence: mandatory
  C:
    version: "~> 2.2"
    presence: mandatory
  Sm:
    version: "~> 1.11.0"
    presence: mandatory
    parameters:
        MTVEC_BASE_ALIGNMENT_DIRECT: {} # Unconstrained
        MTVEC_BASE_ALIGNMENT_VECTORED: {} # Unconstrained
```

becomes:

```yaml
extensions:
  - name: "I"
    version: "~> 2.1"
    presence: mandatory
  - name: "C"
    version: "~> 2.2"
    presence: mandatory
  - name: "Sm"
    version: "~> 1.11.0"
    presence: mandatory
    parameters:
        - name: "MTVEC_BASE_ALIGNMENT_DIRECT": {} # Unconstrained
        - name: "MTVEC_BASE_ALIGNMENT_VECTORED": {} # Unconstrained
```

Here's ERD generated from mc100.yaml as well:
[image]

jalobaba, 2024-11-21:
Up until recently the extensions were an array. @dhower-qc and I (this is James Ball from my personal account) changed it to a hash with the extension name as the key. This was done to support the new $inherit capability so I'm not sure we have much wiggle room here but Derek could elaborate. It had to do with how do you merge hashes versus arrays. The extension names have to be unique already and are suitable as keys (in case we don't end up changing it). Ditto for the parameter names although I don't think those were originally an array.

Kevin, can you elaborate on what problem it creates having these as keys versus arrays? One thing that showed up for me yesterday in helping @dlweaver with his profile creation for RVB23 was that the order that extensions are listed in the profile wouldn't always match the order that they show up in the PDF. Note this isn't an issue for the parameters because they currently don't show up in adoc profiles anyways.

dhower-qc, 2024-11-21:
James gave a good overview. To expand a bit more:

Assume we have profile A1 and then want to inherit A1's extensions in profile A2 _with modifications_. Specifically, let's move the `B` extension from optional to mandatory

Consider the array schema:

```yaml
name: A1
extensions:
- name: "C"
  presence: mandatory
- name: "B"
  presence: optional
```

```yaml
name: A2
extensions:
  $inherit: profile_release/A1.yaml#/extensions
  # how do I find "B" in this array to change optional?
  # I won't know what index it's at, so we'd need a content lookup/match
```

Now consider the object schema:

```yaml
name: A1
extensions:
  C:
    presence: mandatory
  B:
    presence: optional
```

```yaml
name: A2
extensions:
  $inherit: profile_release/A1.yaml#/extensions
  B:
    presence: mandatory
```