riscv/riscv-unified-db #298: Create UDB representation for certification test plan items

james-ball-qualcomm, 2024-11-21:
Just discussed in CSC Test&Models Recommendations WG meeting today. The idea is that we can create certification test plans as a function of a CRDs list of extensions and allowed parameter values if the UDB contains a list of all test plan cover points.

Adding @umershahidengr and @allenjbaum

dhower-qc, 2024-11-22:
What's the deadline for this?

james-ball-qualcomm, 2024-11-22:
Adnan's been talking about these the most lately. Creating test plans (listing coverage points and how they will be covered) is the next logical step after creating CRDs. We already have the MC100 and after @dlweaver gets the RVB23/RVA23 Profiles in place, we can create the AC100 and AC200 CRDs. So, I'd say ~1 month.

james-ball-qualcomm, 2024-11-22:
From the RVCP Overview Slides:

[image]

[image]

closed 2025-05-20 by james-ball-qualcomm