riscv/riscv-unified-db #300: Riscv-opcodes replacement

AFOliveira, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
We have achieved riscv-opcodes source of information in terms of opcodes and variables, but we don't have it's outputs, which are the most valuable.

**Describe the solution you'd like**
We could create a way of generating the outputs that riscv-opcodes uses. From my experince the encoding.c is used for SPIKE and the .go is used by the Golang language.

**Describe alternatives you've considered**
I know there is someone assigned to generating the riscv-opcodes DB, but if we could produce it's outputs, we could propose to replace it entirely and in a short timespan

AFOliveira, 2024-11-22:
cc @dhower-qc @ThinkOpenly

ThinkOpenly, 2024-11-22:
@BrianAnakPintar, did you want to talk about what you are working on here?

BrianAnakPintar, 2024-11-22:
So there are several outputs generated by riscv-opcodes. Currently I have made a script which reads the instructions yaml files from unified db and produced one of the riscv-opcodes' output. I'm currently verifying to make sure the output is the same as riscv-opcodes's outputs.

Initially we tried recreating the original riscv opcodes DB, but soon found out that most of the data doesn't particularly follow a specific format strictly and could contain a lot of "edge cases". So we tried changing our focus into generating the output files instead which seems to be an easier task as they follow a specific format.

In terms of the outputs here's what is generated by riscv-opcodes.
- [x] instr_dict.json (currently verifying its output)
- [ ] encoding.out.h
- [ ] inst.go
- [ ] instr-table.tex
- [ ] priv-instr-table.tex
- [ ] inst.chisel
- [ ] inst.sverilog 
- [ ] inst.rs
- [ ] inst.spinalhdl

Once I'm done verifying the `json` output, I'm planning to see if we can generate `encoding.out.h` and the `inst.go` from the unified db.

dhower-qc, 2024-11-22:
IIRC, some (all?) of those files are created from a script in riscv-opcodes using instr_dict.json as input. Do I have that right? And, if so, can we just run that script from the UDB-generated instr_dict.json?

AFOliveira, 2024-11-25:
This sounds really great!  I would say that if you are going to verify the files generated from the UDB against the riscv-opcodes, there will be some mismatches on the V extensions, since the way riscv-opcodes describes one of the fields, is ambiguous and I tried to make it more concise in the UDB(although I am not still fully sure it is the best approach). However, if you find any other discrepancies, or if you see any way in which the parameters could be better defined, feel free to report that feedback so we can make the correct adjustments :)