riscv/riscv-unified-db #307: Add way for UDB-defined extensions to be identified in extension YAML file and in all artifacts displaying extensions.

james-ball-qualcomm, 2024-11-22:
Currently we have 3 UDB-defined extensions: Sm, Smpmp, and U. We should add some way to indentify these clearly in their extension YAML files (maybe a different specification status) and ensure all artifacts make this clear.

dhower-qc, 2024-11-22:
One could argue that Sm and U are not UDB-defined. Sm1p12, Sm1p13 are referenced by RVI. U is clearly an extension, it just doesn't have a chapter.

Smpmp is definitely new. There is also Smhpm. I suggest we change the state for those to "development"

james-ball-qualcomm, 2025-07-21:
@7908837174

james-ball-qualcomm, 2025-07-22:
The discussion for this issue is largely happening in PR #903.

7908837174, 2025-07-22:
> (riscv-software-src/riscv-unified-db#307)
>
> The discussion for this issue is largely happening in PR #903

Yes