riscv/riscv-unified-db #312: help with getting singularity/apptainer running on arm based Apple macbook running Sequoia 15.1.1

kbroch-rivosinc, 2024-11-24:
## apptainer  read-only error

followed direction here: https://apptainer.org/docs/admin/main/installation.html#mac. basically:

install: 
```
 brew reinstall qemu lima
```

run: 
```
limactl start template://apptainer
limactl shell apptainer
```

any time I try to run something it starts with trying to download something and fails with read-only:

```
kbroch@lima-apptainer:/Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db$ ./do
mkdir: cannot create directory ‘/Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db/.home’: Read-only file system
Fetching container...
mkdir: cannot create directory ‘/Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db/.singularity’: Read-only file system
INFO:    Downloading oras image
6.7.5MiB / 401.7MiB [==>----------------------------------------------------------------------------------------------------------------------------------------] 2 % 2.8 MiB/s 2m21s
```

### debug

cfg says dir is "writable" and dir is user writable:

```
/Users/kbroch/.lima/apptainer/lima.yaml
23:  writable: true
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/p…-initial-cfg:main ⇣11⇡3 !6                                                                          at 06:09:35 PM
❯ rg -A 2 -B 2 "writable" /Users/kbroch/.lima/apptainer/lima.yaml
21-- location: "~"
22-- location: "/tmp/lima"
23:  writable: true
24-containerd:
25-  system: false
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/p…-initial-cfg:main ⇣11⇡3 !6                                                                          at 06:10:04 PM
❯ ls -l /tmp/lima
total 0
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/p…-initial-cfg:main ⇣11⇡3 !6                                                                          at 06:10:12 PM
❯ ls -ld /tmp/lima
drwxrwxr-x  2 kbroch  wheel  64 Nov 23 07:28 /tmp/lima
```

## singularity error stream ID 1; PROTOCOL_ERROR; received from peer

didn't even get as far with some, downloaded the yaml config file: https://raw.githubusercontent.com/sylabs/singularity/main/examples/lima/singularity-ce.yml

but got this: 

```
~/rvi/repos/riscv-software-src/riscv-unified-db on dev/kbroch/p…-initial-cfg:main ⇣11⇡3 !6                                                                          at 06:11:20 PM
❯ limactl start ~/singularity-ce.yml

? Creating an instance "singularity-ce" Proceed with the current configuration
INFO[0001] Starting the instance "singularity-ce" with VM driver "vz"
INFO[0001] Attempting to download the image              arch=aarch64 digest= location="https://repo.almalinux.org/almalinux/9/cloud/aarch64/images/AlmaLinux-9-GenericCloud-latest.aarch64.qcow2"
Downloading the image (AlmaLinux-9-GenericCloud-latest.aarch64.qcow2)
19.07 MiB / 516.31 MiB [->___________________________________] 3.69% 12.96 MiB/sFATA[0003] failed to download "https://repo.almalinux.org/almalinux/9/cloud/aarch64/images/AlmaLinux-9-GenericCloud-latest.aarch64.qcow2": stream error: stream ID 1; PROTOCOL_ERROR; received from peer
```

dhower-qc, 2024-11-24:
Can you try outside your home directory (/Users/kbroch)? I suspect it's related to that. If moving it works, then I'll try to think through a fix.

kbroch-rivosinc, 2024-11-24:
Thanks for quick response. My problem was I hadn't set the `location: "~"` to `writable: true` (just tmp).

kbroch-rivosinc, 2024-11-24:
actually now it's trying to fetch an amd64 image instead of arm64, so trying to work thru this: 
```
kbroch@lima-apptainer:/Users/kbroch/rvi/repos/riscv-software-src/riscv-unified-db$ ./do
Fetching container...
INFO:    Downloading oras image
401.7MiB / 401.7MiB [===========================================================================================================================================================================================================] 100 % 12.9 MiB/s 0s
FATAL:   While pulling image from oci registry: error fetching image to cache: unable to Download Image: could not open image /home/kbroch.linux/.apptainer/cache/oras/tmp_1864917339 for verification: the image's architecture (amd64) could not run on the host's (arm64)
```

dhower-qc, 2024-11-24:
Try to build it locally (./bin/build_container) rather than fetching it.

kbroch-rivosinc, 2024-11-24:
> Try to build it locally (./bin/build_container) rather than fetching it.

This then takes me down the `fakeroot` mode, which relies on linux kernel >= 3.8 https://apptainer.org/admin-docs/master/user_namespace.html?highlight=fakeroot#user-namespace-requirements
and both doc's hint this won't work:

* https://apptainer.org/admin-docs/master/installation.html#installation-on-windows-or-mac
* https://docs.sylabs.io/guides/3.6/admin-guide/installation.html#installation-on-windows-or-mac

Seems like I should either be running containers with root priv's or get off a mac.

dhower-qc, 2024-11-25:
Yes, unfortunately you'll need either root or fakeroot to build the singularity container.

One more option to consider:

There is also a Dockerfile under `.devcontainer` that builds the same image. If you know how to build a docker image without root (I don't...) then that could be an option. If you do that, let me know and we can get the container type an enviornment setting so the scripts under `./bin` pick the right one.

AFOliveira, 2025-03-18:
I think that given recent advancements this is either fixed or worked around with Docker, right?

closed 2025-03-18 by kbroch-rivosinc