riscv/riscv-unified-db #314: Vector Load/Store Segment Instruction mnemonics are missing

BrianAnakPintar, 2024-11-26:
I was recently testing the results by converting the instructions from UDB to riscv-opcodes' format (#300) and one issue I noticed is that UDB seem to be missing several instructions. I am unsure whether there is a way to obtain these from existing instructions in the UDB or if we simply don't have a way to obtain them:

On the left is from UDB and on the right is riscv-opcodes'
[image]

There's a handful of these missing instructions, here's a few of them:
```json
"vloxseg8ei64_v": {
        "encoding": "111011-----------111-----0000111",
        "variable_fields": [
            "rs1",
            "vd",
            "vm",
            "vs2"
        ],
        "extension": [
            "rv_v"
        ],
        "match": "0xec007007",
        "mask": "0xfc00707f"
    },
"vsoxseg2ei16_v": {
        "encoding": "001011-----------101-----0100111",
        "variable_fields": [
            "rs1",
            "vm",
            "vs2",
            "vs3"
        ],
        "extension": [
            "rv_v"
        ],
        "match": "0x2c005027",
        "mask": "0xfc00707f"
    },
"vssseg8e8_v": {
        "encoding": "111010-----------000-----0100111",
        "variable_fields": [
            "rs1",
            "rs2",
            "vm",
            "vs3"
        ],
        "extension": [
            "rv_v"
        ],
        "match": "0xe8000027",
        "mask": "0xfc00707f"
    },
```

AFOliveira, 2024-11-26:
First, thank you so much for the verification.
I was looking to understand issues with the V extension (riscv/riscv-opcodes/issues/322) and your issue helped me to understand this, so thank you @BrianAnakPintar! 
What I now understand is that riscv-opcodes generates this instructions rather then describing them all individually, thus I was not able to fetch them.

Proposed solutions I see for this:

1. Create a back-end to expand instructions just like riscv-opcodes does.
2. Describe them all independently such as @BrianAnakPintar proposed in the beggining of the issue.

I am more inclined towards the 2, since I think it better serves the purpose of having a DB of instructions, but probably the argument to defend 1 is to avoid almost duplicated instructions since they only slightly differ from each other. 

I think doing 2 would be  relatively easy (probably doable until the end of the week).

What do you guys think would be the best approach?

AFOliveira, 2024-11-26:
Additionally, I just realized that .json in riscv-opcode is relatively new and does not exist in the version of riscv-opcodes we have as a submodule here in the UDB, that might be another reason as to why this slipped :). @dhower-qc Do you think it's worth it to PR the update of riscv-opcodes, it's come a long way and I guess @BrianAnakPintar also needs it.

closed 2024-12-02 by dhower-qc