riscv/riscv-unified-db #317: Can "profiles" get a "kind" field, too?

ThinkOpenly, 2024-11-26:
**Is your feature request related to a problem? Please describe.**
I wrote a short Python script to read in all of the YAML files in the project for the purposes of verifying content, like what extensions are referenced, but never defined. The specific YAML files that reference extensions are (at least) instructions, extensions, CSRs, and profiles. The first three have a "kind" field identifying what each YAML represents, but the latter, "profiles", does not. Current "kind" fields are:
```
$ find . -name '*.yaml' -exec grep 'kind:' {} \; | sort -u
kind: certificate class
kind: certificate model
kind: csr
kind: extension
kind: instruction
```

**Describe the solution you'd like**
I'd like to see the YAML files describing profiles also have a kind field like:
```
kind: profile
```

**Describe alternatives you've considered**
It's actually difficult to work around, because the top-level field is a profile name, so not a consistent key that can be checked for existence.

dhower-qc, 2024-12-04:
Yes, that's the plan. The only reason it isn't done now is because @dlweaver is working on RVA23; I didn't want to interfere with the schema until he is ready.

dlweaver, 2024-12-04:
If all it takes is adding a "kind: profile" field to RVB23 and RVA23, I can do that right away.  Just let me know.   (RVB23 is mostly done; probably won't start on RVA23 until early next week)

PR #350 (2024-12-17): Big, Ugly Patch
A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
commit 46285e78 (2024-12-17): A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
closed 2024-12-17 by dhower-qc