riscv/riscv-unified-db #318: [Question] Implicit sfence.vma when satp.MODE is changed from/to Bare

atsushi-shinbo-esoltrinity, 2024-11-27:
**Current value and location**

https://github.com/riscv-software-src/riscv-unified-db/blob/c7a9658554dbb2c627387902a44b8bb0ed605aef/arch/csr/satp.yaml#L46-L47

https://github.com/riscv-software-src/riscv-unified-db/blob/c7a9658554dbb2c627387902a44b8bb0ed605aef/arch/csr/satp.yaml#L64-L65

**Expected value**

I would like to ask for clarification of the above statement.

The privilege spec version 20240411 makes the following two statements regarding changes to satp:
```
11.1.11. Supervisor Address Translation and Protection (satp) Register

Note that writing satp does not imply any ordering constraints between page-table updates and
subsequent address translations, nor does it imply any invalidation of address-translation caches. If
the new address space’s page tables have been modified, or if an ASID is reused, it may be necessary to
execute an SFENCE.VMA instruction (see Section 10.2.1) after, or in some cases before, writing satp.
```
```
11.2.1. Supervisor Memory-Management Fence Instruction

Changing satp.MODE from Bare to other modes and vice versa also takes
effect immediately, without the need to execute an SFENCE.VMA instruction. Likewise, changes to
satp.ASID take effect immediately.
```

- I think `"Note that writing satp does not imply any ordering constraints between page-table updates and subsequent address translations, nor does it imply any invalidation of address-translation caches"` is conflicted with the statements in riscv-unified-db above.

- I interpret this to mean that turning mapping off by switching to bare mode takes effect immediately, i.e. it disables the mapping without a fence, but then switching back to a non-bare mode enables mapping while retaining the state that existed before it was switched to bare.

In general, the software has to execute SFENCE.VMA explicitly when changing satp CSR, i.e. no implicit TLB flush is performed when changing satp.ASID and satp.PPN. However, the RISC-V privilege spec does not describe the implicit SFENCE.VMA when changing satp.MODE from/to Bare.

So, I want to clarify this.

ThinkOpenly, 2024-11-27:
This probably isn't the best place for this question, which seeks clarification of ISA content. This project is really about quantifying that content into a machine-readable database.

I suggest posted to the "ISA-dev" mailing list: https://groups.google.com/u/0/a/groups.riscv.org/g/isa-dev.

atsushi-shinbo-esoltrinity, 2024-12-04:
@dhower-qc

I understood that this repository is about quantifying into a machine-readable database and is not the right place to discuss about RISC-V ISA.

If you know I'd like to ask. 
The following description is committed by you, but where do this description come from ?
https://github.com/riscv-software-src/riscv-unified-db/blame/c7a9658554dbb2c627387902a44b8bb0ed605aef/arch/csr/satp.yaml#L64-L65

AFOliveira, 2025-03-18:
I suppose this is either solved or has been answered elsewhere, I'm closing it just for clearing out issues. If it should still be open, please correct my action.

closed 2025-03-18 by AFOliveira