riscv/riscv-unified-db #319: Incorrect syntax for some instructions' "definedBy" extensions

ThinkOpenly, 2024-11-27:
Some instructions' "definedBy" fields appear to be incorrect, using an open comma-separated list, rather than a subfield refinement with "allOf".
```
./arch/inst/F/fmaxm.s.yaml:definedBy: F, Zfa
./arch/inst/F/froundnx.s.yaml:definedBy: F, Zfa
./arch/inst/F/fround.s.yaml:definedBy: F, Zfa
./arch/inst/F/fltq.s.yaml:definedBy: F, Zfa
./arch/inst/F/fleq.s.yaml:definedBy: F, Zfa
./arch/inst/F/fli.s.yaml:definedBy: F, Zfa
./arch/inst/F/fminm.s.yaml:definedBy: F, Zfa
```

I think each of these should be expanded to:
```
definedBy:
  allOf:
  - F
  - Zfa
```

Although, since Unpriv spec Version 20240411 says:
> The Zfa extension depends on the F extension

maybe only the Zfa extension needs to be called out as the "definedBy" extension?

AFOliveira, 2024-12-04:
This is indeed true and those instructions are missing on the output manual. I'll fix this ASAP. @dhower-qc Do you think  it is worth it to add a test on this to the smoke tests?

dhower-qc, 2024-12-04:
> Do you think it is worth it to add a test on this to the smoke tests?

definitely!!

PR #350 (2024-12-17): Big, Ugly Patch
A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
commit 46285e78 (2024-12-17): A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
closed 2024-12-17 by dhower-qc