riscv/riscv-unified-db #320: Add missing extensions

ThinkOpenly, 2024-11-27:
I wrote a script that looked for references to extensions for which there was no corresponding extension definition. The following extensions are referenced, but have no definition:

- Q
- Smrnmi
- Zalasr
- Zfa
- Zfbfmin
- Zfh
- Zicond
- Zimop
- Zvfbfmin
- Zvfbfwma
- Zvkg

The script also found these, but there are issues open for adding them already:
- Smdbltrp
- Zabha
- Zacas
- Zawrs
- Sdext

(This script could be a good thing to add to CI, I suppose.)

ThinkOpenly, 2024-11-30:
And, my script missed some extensions that are referenced, but not defined. I think adding these completes the set:
- Zbkb
- Zbkc
- Zbkx
- Zbp
- Zca
- Zk
- Zkn
- Zknd
- Zkne
- Zknh
- Zks
- Zksed
- Zksh
- Zvbb
- Zvbc
- Zvkn
- Zvkned
- Zvknha
- Zvknhb
- Zvks
- Zvksed
- Zvksh

james-ball-qualcomm, 2024-12-03:
Ssu64xl is also missing. Defined by RVA20 profile. I'm adding it in #332.

I also will be adding these since they are needed for RVA22:
- Zfh
- Zkn
- Zks

PR #350 (2024-12-17): Big, Ugly Patch
A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
commit 46285e78 (2024-12-17): A Big, Ugly Patch

 * Performs YAML resolution in Python, using lib/yaml_resolver.py
 * Adds many missing extensions
 * Changes Profile schema to match other objects
 * Implements versioning via lib/version.rb instead of Gem::Version/Requirement
 * Refactors ArchDef into Specification/Config
 * Simplifies ArchDef/Specification by exploiting common format for object schemas
 * New, simplified schema for configs
 * Adds schemas for profile (some are still placeholders)
 * Parallelizes GitHub regression test

fixes #336
fixes #320
fixes #319
fixes #317
fixes #299
closed 2024-12-17 by dhower-qc